DSheirer/sdrtrunk #635: nooElectric NESDR SMART v4 not recognized Windows 10 (latest patch levels)

8F27E956, 2020-01-09:
New nooElec NESDR SMART v4 dongle. V0.4.0-beta.2, cut/paste from the DOS boot window. Windows 10, 64 bit, latest patch levels. Tried various USB drivers with same outcome -- Tuner not recognized.

I've searched here for solutions and see other cases (similar) but they either suggest fixed by a version earlier than 040-b2 or haven't been resolved in public.

Sorry if this is covered ground and I simply didn't see it.

SPECS
SKU | 100700
RTL2832U
Tuner | R820T2
25MHz - 1750MHz
---

Boot screen scrape

15:15:16.776 INFO  i.g.d.log.ApplicationLog - Application Log File: C:\Users\Scotty\SDRTrunk\logs\sdrtrunk_app.log  [18MB/510MB 3%]
15:15:16.799 INFO  i.g.d.log.ApplicationLog -   [19MB/510MB 3%]
15:15:16.805 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [19MB/510MB 3%]
15:15:16.805 INFO  i.g.d.log.ApplicationLog - **** sdrtrunk: a trunked radio and digital decoding application ***  [19MB/510MB 3%]
15:15:16.805 INFO  i.g.d.log.ApplicationLog - ****  website: https://github.com/dsheirer/sdrtrunk             ***  [19MB/510MB 3%]
15:15:16.806 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [19MB/510MB 3%]
15:15:16.806 INFO  i.g.d.log.ApplicationLog - Memory Logging Format: [Used/Allocated PercentUsed%]  [19MB/510MB 3%]
15:15:16.806 INFO  i.g.d.log.ApplicationLog - Host OS Name:          Windows 10  [19MB/510MB 3%]
15:15:16.807 INFO  i.g.d.log.ApplicationLog - Host OS Arch:          amd64  [19MB/510MB 3%]
15:15:16.807 INFO  i.g.d.log.ApplicationLog - Host OS Version:       10.0  [19MB/510MB 3%]
15:15:16.807 INFO  i.g.d.log.ApplicationLog - Host CPU Cores:        8  [19MB/510MB 3%]
15:15:16.807 INFO  i.g.d.log.ApplicationLog - Host Max Java Memory:  7 GB  [19MB/510MB 3%]
15:15:16.808 INFO  i.g.d.log.ApplicationLog - Storage Directories:  [19MB/510MB 3%]
15:15:16.809 INFO  i.g.d.log.ApplicationLog -  Application Root: C:\Users\Scotty\SDRTrunk  [19MB/510MB 3%]
15:15:16.809 INFO  i.g.d.log.ApplicationLog -  Application Log:  C:\Users\Scotty\SDRTrunk\logs  [19MB/510MB 3%]
15:15:16.810 INFO  i.g.d.log.ApplicationLog -  Event Log:        C:\Users\Scotty\SDRTrunk\event_logs  [19MB/510MB 3%]
15:15:16.811 INFO  i.g.d.log.ApplicationLog -  Playlist:         C:\Users\Scotty\SDRTrunk\playlist  [19MB/510MB 3%]
15:15:16.813 INFO  i.g.d.log.ApplicationLog -  Recordings:       C:\Users\Scotty\SDRTrunk\recordings  [19MB/510MB 3%]
15:15:16.816 INFO  i.g.d.util.ThreadPool - Application thread pool created with [8] threads  [19MB/510MB 3%]
15:15:16.816 INFO  i.g.dsheirer.gui.SDRTrunk - Home path: C:\Users\Scotty\SDRTrunk  [19MB/510MB 3%]
15:15:16.819 INFO  i.g.d.p.SystemProperties - SystemProperties - loaded [C:\Users\Scotty\SDRTrunk\SDRTrunk.properties]  [19MB/510MB 3%]
15:15:16.820 INFO  i.g.d.p.SystemProperties - SystemProperties - application properties loaded [C:\Users\Scotty\SDRTrunk\SDRTrunk.properties]  [19MB/510MB 3%]
15:15:16.839 INFO  i.g.d.s.SettingsManager - SettingsManager - loading settings file [C:\Users\Scotty\SDRTrunk\settings\settings.xml]  [21MB/510MB 4%]
15:15:17.161 INFO  i.g.d.s.m.MixerManager - loading system mixer devices
        [LOADED]     Output: Primary Sound Driver CHANNELS: [MONO, STEREO]
        [LOADED]     Output: BenQ GW2470 (Intel(R) Display Audio) CHANNELS: [MONO, STEREO]
        [LOADED]     Output: BenQ GW2470 (Intel(R) Display Audio) CHANNELS: [MONO, STEREO]
  [11MB/34MB 32%]
15:15:17.162 INFO  i.g.d.s.r.RecordingSourceManager - RecordingSourceManager - discovered [0] recording configurations  [11MB/34MB 32%]
15:15:17.944 INFO  i.g.d.s.t.TunerManager - LibUSB API Version: 16777478  [13MB/34MB 39%]
15:15:17.945 INFO  i.g.d.s.t.TunerManager - LibUSB Version: 1.0.22.11312  [14MB/34MB 41%]
15:15:17.969 INFO  i.g.d.s.t.TunerManager - discovered [9] attached USB devices  [14MB/34MB 41%]
15:15:17.973 INFO  i.g.d.s.t.TunerManager - usb device [258A:1006] NOT LOADED: Unknown Device  [14MB/34MB 41%]
15:15:17.973 INFO  i.g.d.s.t.TunerManager - usb device [0BDA:0411] NOT LOADED: Unknown Device  [14MB/34MB 41%]
15:15:17.974 INFO  i.g.d.s.t.TunerManager - usb device [0BDA:5411] NOT LOADED: Unknown Device  [14MB/34MB 41%]
15:15:17.978 ERROR i.g.d.s.t.r.RTL2832TunerController - error while determining tuner type  [14MB/34MB 41%]
io.github.dsheirer.source.SourceException: couldn't claim usb interface [LIBUSB_ERROR_NOT_SUPPORTED]
        at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.claimInterface(RTL2832TunerController.java:201)
        at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.identifyTunerType(RTL2832TunerController.java:350)
        at io.github.dsheirer.source.tuner.TunerManager.initRTL2832Tuner(TunerManager.java:383)
        at io.github.dsheirer.source.tuner.TunerManager.initTuner(TunerManager.java:225)
        at io.github.dsheirer.source.tuner.TunerManager.initTuners(TunerManager.java:140)
        at io.github.dsheirer.source.tuner.TunerManager.<init>(TunerManager.java:72)
        at io.github.dsheirer.source.SourceManager.<init>(SourceManager.java:48)
        at io.github.dsheirer.gui.SDRTrunk.<init>(SDRTrunk.java:187)
        at io.github.dsheirer.gui.SDRTrunk.main(SDRTrunk.java:670)
15:15:17.990 INFO  i.g.d.s.t.TunerManager - usb device [0BDA:2838] NOT LOADED: SDRTRunk doesn't currently support RTL2832 Dongle with [UNKNOWN] tuner for tuner class[USB Tuner:RTL2832_VARIOUS Vendor:RTL2832 Device:SDR Address:0BDA:2838]  [14MB/34MB 41%]
15:15:17.990 INFO  i.g.d.s.t.TunerManager - usb device [8087:0A2B] NOT LOADED: Unknown Device  [14MB/34MB 41%]
15:15:17.991 INFO  i.g.d.s.t.TunerManager - usb device [047D:1020] NOT LOADED: Unknown Device  [14MB/34MB 41%]
15:15:17.991 INFO  i.g.d.s.t.TunerManager - usb device [0BDA:5411] NOT LOADED: Unknown Device  [14MB/34MB 41%]
15:15:17.992 INFO  i.g.d.s.t.TunerManager - usb device [8086:9D2F] NOT LOADED: Unknown Device  [14MB/34MB 41%]
15:15:17.993 INFO  i.g.d.s.t.TunerManager - usb device [0BDA:0411] NOT LOADED: Unknown Device  [14MB/34MB 41%]
15:15:18.039 INFO  i.g.d.a.AudioPacketManager - Audio packet manager started  [16MB/34MB 49%]
15:15:18.658 INFO  i.g.d.icon.IconManager - loading icons file [C:\Users\Scotty\SDRTrunk\settings\icons.xml]  [10MB/34MB 32%]
15:15:18.825 INFO  i.g.d.p.PlaylistManager - Loading playlist file [C:\Users\Scotty\SDRTrunk\playlist\default.xml]  [15MB/34MB 44%]
15:15:18.861 INFO  i.g.dsheirer.gui.SDRTrunk - starting main application gui  [17MB/34MB 50%]

Help greatly appreciated.

DSheirer, 2020-01-10:
@8F27E956 The log indicates that something else has claimed the tuner and therefore sdrtrunk can't claim it.

Are you able to use it with another application like sdrconsole or sdr# ??

8F27E956, 2020-01-12:
Having /just/ inserted the dongle the first time and proceeded with the drivers updates (ZaDig), and being at the /very/ beginning of my adventure with sdrTrunk and sdr dongles, I cannot imagine what else would have/could have claimed the tuner.

That said, turning OFF the filters in ZaDig and trolling the list, one discovers that there are two (2) dongle devices, suffixed (0) and (1). Only the (1) shows in the filtered list. The (1) was the "target" I used for ZaDig driver installs. Turns out sdrTrunk finds and operates the (0), which per above sentence, didn't have the needed ZaDig treatment/drivers.

[Re-] Target the (0) entry, and everybody is happy.

FYI, in the Windows 10 device manager (not ZaDig), there is only a single USB dongle device -- thus I don't know where/how the (0) and (1) are happening.

Oddly, I have seen this before with android smartphones and USB drivers, but it was a couple of years a go and many versions of windows 10 past. 

While I remain confused ... I am now operational.

Enjoying day two with sdrTrunk. Thanks for an interesting piece of software.

Likely, many more questions will follow.

Thanks & cheers,

DSheirer, 2020-01-12:
Sounds good.  I'll close this ticket then.

closed 2020-01-12 by DSheirer