DSheirer/sdrtrunk #652: USBMasterProcessor Seriving Thread Revisit Rate Should Be A User Configurable Item

DSheirer, 2020-02-04:
Either expose the Processor thread scheduled revisit rate setting to the user preferences so that user's with many tuners can adjust it, or possibly change the rate adaptively based on the number of tuners that are being used. 

https://groups.google.com/forum/#!topic/sdrtrunk/87oGoT5O3Tc

closed 2020-02-09 by DSheirer

ImagoTrigger, 2020-02-14:
very nice. i get the warning but it seems to work smooth.

6Mhz airspy mini
2.560mhz r820t
2.560mhz r820t

^--  on one bus  

below image shows all tuners locked in use

[image]

👍

DSheirer, 2020-02-14:
@ImagoTrigger when I use my airspy at full 10MHz rate along with a 2.4MHz RTL dongle, on one bus it seems to work fine, but on the other, I get lots of dropped packets and choppy audio.  It's very visible in the spectral display also, with lots of horizontal lines where the data losses occur.