DSheirer/sdrtrunk #661: doesn't detect both of my identical rtl dongles

HuRRiK41N, 2020-02-18:
beta version 4 only detects one dongle. I think they have the same serial number 000000001, but it shows up as two separate dongles in sdr#

hans109h, 2020-02-18:
Change the serial number so they each are unique.  Google the software requirement.

DSheirer, 2020-02-18:
@HuRRiK41N does the second tuner show up in the tuner's tab?  If not, are you seeing any error messages in the console window when the application starts up?

As @hans109h mentioned, you'll need unique serial numbers for each dongle for sdrtrunk to configure each of them properly.

HuRRiK41N, 2020-02-20:
ive since fixed it by trying the port on the opposite side of my laptop

DSheirer, 2020-02-21:
Very good.  I'll close this ticket.

closed 2020-02-21 by DSheirer