DSheirer/sdrtrunk #662: Trying to load Preferences throws exception

camcabbit, 2020-02-20:
Every time I try to open the Preferences (View/Preferences), no window pops up and I get an exception in the command line window.

Windows 10 x64 Home
Intel Core i7-6500U @ 2.5 GHz
16GB ram
Oracle Java 8 update 241 (both x86 and x64 are installed)

Any clue what's going on? I'm running it from the sdrtrunk.bat file, if that helps.

[log.txt](https://github.com/DSheirer/sdrtrunk/files/4232034/log.txt)

DSheirer, 2020-02-21:
@camcabbit Are you using one of the released versions of sdrtrunk, or are you compiling the code yourself?  

If you are compiling the code yourself and using your local JDK, then check to make sure your JDK has the JavaFX libraries.  Some of the newer JDK versions no longer include the JavaFX libraries.

The pre-compiled release versions that I provide on the website have Java included along with all of the JavaFX libraries.

camcabbit, 2020-02-21:
I'm using the pre-compiled release contained in sdr-trunk-0.4.0-beta.4-windows-x86_64.zip on the releases page.

DSheirer, 2020-02-21:
@camcabbit After searching google, this seems to be a known issue that points to Java being misconfigured on the end user machine.  A JDK bug link is here: https://bugs.openjdk.java.net/browse/JDK-8231015

The issue seems to be that the sdrtrunk embedded Java is getting confused and using some of the Java libraries from one of your local installed Java versions.  I suspect that since sdrtrunk is using 64-bit java, someone it's picking up the 32-bit Java DLL libraries on your local Java installation.

Recommendation:
1. Uninstall the x86 version of Java.  You should only have 1 active installation of java on your machine and since you have more than 4GB of ram, you should be using the 64-bit version of java.  You can simply modify your PATH environment variable to remove the x86 version off of the system path and try that.

camcabbit, 2020-02-22:
Uninstalling Java 32bit worked. I also cleared the path of any Java 32bit paths. Thanks for the assistance.

closed 2020-02-22 by camcabbit