DSheirer/sdrtrunk #677: Is there a recording limit?

LordDonkey, 2020-03-19:
I'm asking because there was a 67 second transmission and it's not showing up in the recordings folder. Also, some transmissions aren't being recorded also or they are overlapped in other recordings.

Some of the metadata isn't making it over the stream, it's showing as TO:UNKNOWN FROM:UNKNOWN.

DSheirer, 2020-03-20:
With the latest commits to master, the recording limit should be 1 second.  Were you running a build from master, or the 0.4.0 beta 5 release?

For the recording overlap issue ... which Decoder?  Are you gettting good/solid decodes when you're seeing this issue?

DSheirer, 2020-03-20:
I forgot to mention ... it will chop audio segments at 60 seconds.  Recordings should be a maximumof 60 seconds.  Playback of the audio will appear seamless because I link continuous audio segements together so that playback doesn't get interrupted.

LordDonkey, 2020-03-20:
It was the beta 5 build from yesterday's master. The issue exists with today's compile from the master as well. The system I'm on has a Phase I control channel and Phase II voice channels. The decodes are good, no errors, strong signal. Some calls are not being recorded. There's also recordings from separate RIDs, but on the same talkgroup, that are being put in the same file.

LordDonkey, 2020-03-21:
The latest master build fixed most if this issue. Recordings are being made and the stream appears to be working properly, minus the occasional TO:UKNOWN FROM:UNKNOWN in the metadata.

DSheirer, 2020-03-22:
@LordDonkey I merge another branch into master yesterday afternoon (2pm EST) with some Phase II changes.  Did you happen to rebuild and run these latest changes?

There were some bugs in the code that may have contributed to the recordings being cutoff like you were seeing.

LordDonkey, 2020-03-22:
Yes sir, the audio cutoff issue is now fixed.
Shoutcast V2 Stream metadata still occasionally shows as TO:UNKNOWN FROM:UNKNOWN. [Here](http://s2.free-shoutcast.com:18184/played.html?sid=1) is a link to the stream history.

DSheirer, 2020-03-22:
Others are reporting the same issue for Icecast as well.  I suspect these are empty audio segments, but I'll have to do some testing to confirm and figure out a way to stop them from happening.

PR #704 (2020-03-28): #677 Streaming Audio Metadata UNKNOWN/UNKNOWN
Resolves #677 Streaming Audio Metadata is showing UNKNOWN/UNKNOWN
commit 5e62f1b8 (2020-03-28): Resolves #677 Streaming Audio Metadata UNKNOWN/UNKNOWN (#704)
closed 2020-03-28 by DSheirer