DSheirer/sdrtrunk #690: Streaming to Icecast 2.4.4 Server

DSheirer, 2020-03-22:
Test connections to Icecast 2.4.4 server.

Current code is only tested against 2.4.2 servers.

dlsauer, 2020-03-22:
Testing with Harris system audio to 2.4.4 on Kubuntu 14.04!

Loud and clear

Connects.. fine..

Stream PERFECT!

METATAGS  FAIL -FROM UNKNOWN to UNKNOWN. :( :( :(

hans109h, 2020-03-22:
Same here.  just to clarify, the metatags being sent to icecast (or at least icecast is displaying) is "TO: UNKNOWN FROM: UNKNOWN" for all clips.

sdevine01, 2020-04-01:
I seem to have it working with win32 Icecast 2.4.4.  Metadata also seems to be working.

DSheirer, 2020-04-01:
Ok.  I think with the metadata fix in 0.4.0 beta 8 this ticket is resolved.  Thanks to everyone for the feedback.

closed 2020-04-01 by DSheirer