DSheirer/sdrtrunk #694: NPE at Playback Preferences

tateconcepts, 2020-03-22:
Hmm. Not sure what happened. I just installed the latest beta as of today and also are using the latest jbme 1.0.2 also just committed today. I was able to start this the first time using a single NESDR Smart however I could not get any decent reception to test NOAA on. Once I moved the antenna and rebooted the VM for required Windows updates, this NPE occurred. 

This is a Windows Server 2012 R2 VM on VMware guest
4 vCPU
8GB RAM
80 GB DISK
Using Liberica JDK 8 

C:\SDRTrunk\bin>sdr-trunk.bat
14:26:50.113 INFO  i.g.d.p.d.DirectoryPreference - Created directory [C
dministrator\SDRTrunk]  [8MB/128MB 6%]
14:26:50.113 INFO  i.g.d.p.d.DirectoryPreference - Created directory [C
dministrator\SDRTrunk\logs]  [8MB/128MB 6%]
14:26:50.113 INFO  i.g.d.log.ApplicationLog - Application Log File: C:\
inistrator\SDRTrunk\logs\sdrtrunk_app.log  [8MB/128MB 6%]
14:26:50.128 INFO  i.g.d.log.ApplicationLog -   [9MB/128MB 7%]
14:26:50.144 INFO  i.g.d.log.ApplicationLog - *************************
*********************************  [9MB/128MB 7%]
14:26:50.144 INFO  i.g.d.log.ApplicationLog - **** sdrtrunk: a trunked
 digital decoding application ***  [9MB/128MB 7%]
14:26:50.144 INFO  i.g.d.log.ApplicationLog - ****  website: https://gi
dsheirer/sdrtrunk             ***  [9MB/128MB 7%]
14:26:50.144 INFO  i.g.d.log.ApplicationLog - *************************
*********************************  [9MB/128MB 7%]
14:26:50.144 INFO  i.g.d.log.ApplicationLog - Memory Logging Format: [U
ated PercentUsed%]  [9MB/128MB 7%]
14:26:50.144 INFO  i.g.d.log.ApplicationLog - Host OS Name:          Wi
ver 2012 R2  [9MB/128MB 7%]
14:26:50.144 INFO  i.g.d.log.ApplicationLog - Host OS Arch:          am
/128MB 7%]
14:26:50.144 INFO  i.g.d.log.ApplicationLog - Host OS Version:       6.
28MB 7%]
14:26:50.144 INFO  i.g.d.log.ApplicationLog - Host CPU Cores:        4
MB 7%]
14:26:50.144 INFO  i.g.d.log.ApplicationLog - Host Max Java Memory:  2
128MB 7%]
14:26:50.144 INFO  i.g.d.log.ApplicationLog - Storage Directories:  [9M
%]
14:26:50.144 INFO  i.g.d.log.ApplicationLog -  Application Root: C:\Use
strator\SDRTrunk  [9MB/128MB 7%]
14:26:50.144 INFO  i.g.d.log.ApplicationLog -  Application Log:  C:\Use
strator\SDRTrunk\logs  [9MB/128MB 7%]
14:26:50.144 INFO  i.g.d.p.d.DirectoryPreference - Created directory [C
dministrator\SDRTrunk\event_logs]  [9MB/128MB 7%]
14:26:50.144 INFO  i.g.d.log.ApplicationLog -  Event Log:        C:\Use
strator\SDRTrunk\event_logs  [9MB/128MB 7%]
14:26:50.144 INFO  i.g.d.p.d.DirectoryPreference - Created directory [C
dministrator\SDRTrunk\playlist]  [9MB/128MB 7%]
14:26:50.144 INFO  i.g.d.log.ApplicationLog -  Playlist:         C:\Use
strator\SDRTrunk\playlist  [9MB/128MB 7%]
14:26:50.144 INFO  i.g.d.p.d.DirectoryPreference - Created directory [C
dministrator\SDRTrunk\recordings]  [9MB/128MB 7%]
14:26:50.144 INFO  i.g.d.log.ApplicationLog -  Recordings:       C:\Use
strator\SDRTrunk\recordings  [9MB/128MB 7%]
14:26:50.160 INFO  i.g.d.util.ThreadPool - Application thread pool crea
[4] threads  [9MB/128MB 7%]
14:26:50.160 INFO  i.g.dsheirer.gui.SDRTrunk - Home path: C:\Users\Admi
\SDRTrunk  [10MB/128MB 8%]
14:26:50.160 INFO  i.g.dsheirer.gui.SDRTrunk - SDRTrunk - creating appl
roperties file [C:\Users\Administrator\SDRTrunk\SDRTrunk.properties]  [
B 8%]
14:26:50.160 INFO  i.g.d.p.SystemProperties - SystemProperties - loaded
s\Administrator\SDRTrunk\SDRTrunk.properties]  [10MB/128MB 8%]
14:26:50.160 INFO  i.g.d.p.SystemProperties - SystemProperties - applic
perties loaded [C:\Users\Administrator\SDRTrunk\SDRTrunk.properties]  [
B 8%]
14:26:50.332 INFO  i.g.d.s.SettingsManager - SettingsManager - settings
 exist [C:\Users\Administrator\SDRTrunk\settings\settings.xml]  [7MB/12
14:26:50.347 INFO  i.g.d.s.r.RecordingSourceManager - RecordingSourceMa
iscovered [0] recording configurations  [8MB/128MB 6%]
14:26:50.363 INFO  i.g.d.s.t.TunerManager - LibUSB API Version: 1677747
28MB 7%]
14:26:50.363 INFO  i.g.d.s.t.TunerManager - LibUSB Version: 1.0.22.1131
28MB 7%]
14:26:50.378 INFO  i.g.d.s.t.TunerManager - discovered [6] attached USB
 [9MB/128MB 7%]
14:26:50.378 INFO  i.g.d.s.t.TunerManager - USB Bus [1] Device [0E0F:00
evice  [9MB/128MB 7%]
14:26:50.378 INFO  i.g.d.s.t.TunerManager - USB Bus [1] Device [0E0F:00
evice  [9MB/128MB 7%]
14:26:51.911 INFO  i.g.d.s.t.TunerManager - USB Bus [1] Device [0BDA:28
D: RTL2832 SDR/R820T 00000001 Max Rate:38400000 bps  [12MB/128MB 9%]
14:26:51.911 INFO  i.g.d.s.t.TunerManager - USB Bus [1] Device [2040:72
wn Device - Class 0  [12MB/128MB 9%]
14:26:51.911 INFO  i.g.d.s.t.TunerManager - USB Bus [1] Device [0E0F:00
wn Device - Class 0  [12MB/128MB 9%]
14:26:51.911 INFO  i.g.d.s.t.TunerManager - USB Bus [1] Device [15AD:07
evice  [12MB/128MB 9%]
14:26:51.911 INFO  i.g.d.s.t.TunerManager - ---------------------------
-------------------------  [12MB/128MB 9%]
14:26:51.911 INFO  i.g.d.s.t.TunerManager - USB Bus - Potential Maximum
es  [12MB/128MB 9%]
14:26:51.926 INFO  i.g.d.s.t.TunerManager - USB Bus [1] Rate [38400000]
 second  [12MB/128MB 9%]
Exception in thread "main" java.lang.NullPointerException
        at io.github.dsheirer.preference.playback.PlaybackPreference.ge
nnelConfiguration(PlaybackPreference.java:353)
        at io.github.dsheirer.audio.playback.AudioPlaybackManager.<init
aybackManager.java:83)
        at io.github.dsheirer.gui.SDRTrunk.<init>(SDRTrunk.java:195)
        at io.github.dsheirer.gui.SDRTrunk.main(SDRTrunk.java:668)

tateconcepts, 2020-03-22:
I had to read this NPE again and thought, what could be "null" immediately for audio playback?

I looked and Windows had disabled the audio manager with the last update. Please ensure that service is running and you have audio in your VM. Once you do, the issue will be resolved.

DSheirer, 2020-03-23:
This looks to be an error when there's no audio devices.  I'm reopening to fix the code to handle this situtation.

PR #695 (2020-03-23): #694 NPE When No System Audio Devices Available
Resolves #694.  Addresses null pointer exception when no system audio devices are available.
commit 41df207b (2020-03-23): Resolves #694.  Address null pointer when no system audio devices are available. (#695)
closed 2020-03-23 by DSheirer