DSheirer/sdrtrunk #717: Investigate WebHook Feature

DSheirer, 2020-03-29:
Investigate exposing webhook event notification service for decode events. 

Interfaces could include:
1. Publish all events for a channel
2. Publish events via alias actions

References:
https://developer.github.com/webhooks/
https://zapier.com/blog/what-are-webhooks/
https://gist.github.com/jagrosh/5b1761213e33fc5b54ec7f6379034a22

davidmckenzie, 2020-03-29:
With the volume of events adding a WebSockets interface might be wiser. Not sure about libs but I use socket.io in node, and have seen [Pusher](https://pusher.com/channels) used nicely.

ZacharyDuBois, 2020-04-19:
Going with what @davidmckenzie said. WebSockets would probably be much more reliable.