DSheirer/sdrtrunk #718: Reception Priority in the "Now Playing" window

ah1102, 2020-03-29:
Could you add the values ​​of the priority of the reception or its absence in the "Now running" window. It would be convenient to see which priority groups are displayed. Otherwise, not entirely informative. The Now Running window displays the currently accepted group, but I don’t remember its priority. Maybe even add a quick change in priority.

In my country, in one trunk (on 1 site) there is a huge number of services, and in this case I would like to have a tool for quickly assessing the priority of reception and changing it. What do you think?

[image]

DSheirer, 2020-04-06:
@ah1102 There's a new feature I'm going to be adding that will show a list of the most recent calls in time order.  I think that would be a more appropriate place for the priority information and how you would want to use it.

The 'Now Playing' tab is designed more for showing which channels are currently decoding and it changes so rapidly (sometimes) that I think the priority detail would get lost.

ah1102, 2020-04-06:
Yes it would be nice