DSheirer/sdrtrunk #720: APCO-25 "Talkgroup range" does not work

ah1102, 2020-03-30:
APCO-25 "Talkgroup range" does not work.  SDRTrunk software version 0.4.0 b8

[image]

[image]

DSheirer, 2020-03-30:
Check that the talkgroup range 1301-1302 doesn't overlap with any other alias talkgroup ranges that you've defined.

When there is overlap between talkgroup ranges, sdrtunk will use the first one that it finds.

DSheirer, 2020-03-30:
Which part is not working?  Is it not blocking the audio from playback?

ah1102, 2020-03-30:
Overridden only by the blacklist group that I created to block unknown IDs

[image]

The same thing happened with another group, which had to be divided into separate IDs.

[image]

> Which part is not working? Is it not blocking the audio from playback?

All IDs in the Talkgroup range are blocked. No audio output when monitoring priority is enabled.

DSheirer, 2020-03-31:
If you have two or more talkgroup ranges that overlap, the software has no way of knowing which one it should use, so it picks the first one that it finds.

If the first one that it finds is your 0-65535 blacklist, then the call will not be monitored.

If you want to define a talkgroup range 1301 - 1302 to not be part of the blacklist, then you would have to change the 0-65535 blacklist to have two separate talkgroup ranges like this:

0 - 1300
1303 - 65535

This would allow another alias to use the range 1301-1302 and not conflict with your blacklist alias.

closed 2020-04-01 by DSheirer