DSheirer/sdrtrunk #735: Hackrf fix not working in windows

49handyman, 2020-04-09:
moved the sdrtrunk directory
cloned the current master & also tried beta.8

doesnt seem to work on Windows 10 but it worked on linux

let me know if i should try sometime different

thanks for the fix

DSheirer, 2020-04-09:
@49handyman Have you tried the 0.4.0 release (instead of building it yourself) for Windows to verify that there is an issue?

49handyman, 2020-04-10:
that worked thanks.

can you let me know why that might happen on my windows machine. 

i use gradlew clean build if that right

the master has the fix right?

DSheirer, 2020-04-11:
Yes, the source code master branch is the same.  I doesn't matter which OS you're using.

Make sure the source code is up-to-date.  Are you using git or are you just downloadin and unzipping the entire master branch and then compiling?

closed 2020-04-14 by DSheirer