DSheirer/sdrtrunk #736: Redesign the Spectral Display in JavaFX

DSheirer, 2020-04-09:
This ticket captures requirements for the redesign of the Spectral and Waterfall displays in JavaFX.

-Channel Overlay colors should be synchronized to Channel State colors in ChannelMetadata panel.

ZacharyDuBois, 2020-04-19:
I wouldn't tightly couple the spectral display to Radio Reference (or make it an option). I've noticed a few systems on Radio Reference that don't have helpful information or up-to-date information. I would say for P25 control channels that advertise their frequencies, the spectral display should automagically show the available channels, etc. Another thing would be for those using multiple SDRs to cover one trunked site, have the spectral display merge the output of all the attached SDRs (assuming the user has enough bandwidth to cover all the frequencies in the site). Love the idea of channel overlays to match the channel state!

DSheirer, 2020-10-15:
Request for user to assign custom color to each channel to use when the channel is displayed in the spectral display.

closed 2026-11-13 by DSheirer