DSheirer/sdrtrunk #737: compatibility with R&S EB500 receiver

Mahmoudshakra, 2020-04-09:
please any advise to make the source R&S EB500 receiver

DSheirer, 2020-04-09:
I don't have access to an EB500 receiver, so I won't be able to add support for it.

Mahmoudshakra, 2020-04-15:
Hi Dr. D sheirer

hope this email finds you well , i want to ask another question , if the
USRP compatible with SDR Trunk or not ??
kind regards,

Eng.  Mahmoud M.Shakra

DSheirer, 2020-04-17:
@Mahmoudshakra No, the USRP tuner drivers are not currently supported.

whelgeson, 2020-06-23:
I have a [commit](https://github.com/whelgeson/sdrtrunk/commit/9d2900d00875c31e48c6f195309ea19220cfd3a3) that implements USRP support through a UHD wrapper called [uhd-java] (https://github.com/radiowitness/uhd-java). I wrote this for LimeSDR support for full 55Mhz tuning, which is working but it takes a lot of horsepower to tune at full bandwidth.

I never PR'd this because IMO it's rather brittle and at this point I think master has diverged enough that it wouldn't be a clean PR anyway. If someone wants to pick it up and run with it, please do!