DSheirer/sdrtrunk #742: hackrf too close to dc spike loosing control freq.

49handyman, 2020-04-18:
using 10Mhz and 12Mhz bw it keeps jumping too close to DC spike and loosing control channel (100% gone). moving the tuner freq manually brought it back immediately. but it would recalculate in just a few seconds.

i changed offset to 10000  and still happened. But 20000 it worked fine. is that in Hz?

public static final int DC_SPIKE_AVOID_BUFFER = 20000;

my control freq is 853.775Mhz I think it was calculating 853.750Mhz

it was so close i couldn't hardly distinguish it from spike really. zooming in you could barely see signal.

not sure if you've seen this on yours. i'm reviewing the tuner calculations for center freqs, but if you have any hints for me to look at that will help to understand id appreciate it.

DSheirer, 2020-04-18:
The avoid buffer is in hertz.  Simply increase that value to create a wider buffer about the DC spike.

49handyman, 2020-04-19:
worked fine thanks

closed 2020-04-19 by 49handyman