DSheirer/sdrtrunk #748: Streaming ShoutCast to myradiostream.com ...

StaRipper, 2020-04-20:
Hello,
I'm trying to cast to https://myradiostream.com ...
It's a ShoutCast v.1.9.8 server...
The app connecting and immediately disconnecting.
Is there a log for the streaming section?

DSheirer, 2020-04-20:
On the File menu, there's a Logs & Recording menu item that should open a file browser to the directory.  Look in the application log to see if there were any errors posted.

StaRipper, 2020-04-21:
i checked it already.
there is only a line that says "status: Disconnected (143MB/174MB)"
no other reason specified.
with Butt client and same credentials i'm successfully streaming.

StaRipper, 2020-05-10:
Hello,
Just tried on 0.50 Alpha 1. still same issue. cannot stream to shoutcast server.