DSheirer/sdrtrunk #750: Feature Request: One Click Mute and/or Follow

MikonosII, 2020-04-21:
It would be nice to be able to with a single click to be able to follow (mute all others) or mute a talk group as they are on a call.

mart6723, 2020-06-05:
Agreed. For example: if you hear a call you're interested in, you can isolate the talk-group and only hear that TG until released. 
+1

kazoni, 2020-06-17:
Adding support to this idea.  Right click -> Channel ->  Mute isn't fast enough when it gets busy and you have something specific you want to listen to, especially when they don't necessarily stay on the same speaker.  

Perhaps having a timeout duration that would keep the TG on which ever channel it first lands on until X seconds after the last transmission.  I don't know how code intensive that would be.

cmartus, 2020-06-17:
Yeah I thought about that as well. Maybe you could treat it almost like a special playlist. So I know that I want to hear police on left and fire on right or just police mono. But it would have to be something you could quick toggle. Favorites list? I forked this repo and started looking at the playlist code. 

Also, #753 is also in the same ballpark.

Cagier, 2021-10-07:
Along the same lines, there is a "Master" mute button but to mute an individual channel you have to right-click and select which channel to mute.  Also having an individual Mute button for each (left/right) channel would be much quicker and intuitive to use.  (Or a "mute until this transmission is finished" option to temporarily mute.)

MikeRAus, 2023-02-10:
To complement this, I would like an UNMUTE ALL Button e.g. press the Space Bar to override all Alias List Listen settings and instead, listen to every channel that comes on.

Useful when it's very quiet on a System.

MikonosII, 2024-02-11:
@DSheirer any thoughts on adding this or #753?