DSheirer/sdrtrunk #755: Feature Request: Signal strength

ryanthomp, 2020-04-23:
On Unitrunker (which I used in conjunction with DSD in the past) there was a feature where it would show a number and a bar graph for the signal "health" (0-100). I am not sure if this was actually useful or even accurate but I often will find myself in SDRTrunk I have to guess whether or not I am getting a good reception from the control frequency and it would be nice to have some sort of benchmark on how good of reception I am getting.

[image]
[image]

DSheirer, 2020-04-24:
@ryanthomp I have wanted to add a signal view tab that would allow you to click on the channel in the now playing window and in addition to events and messages, there would be a third tab that let's you see information about the channel/signal.  There could be things like a channel spectral display and symbol plots, etc. in this tab.

ryanthomp, 2020-04-24:
Is what you described going to be implemented with #150

dlsauer, 2020-05-25:
> @ryanthomp I have wanted to add a signal view tab that would allow you to click on the channel in the now playing window and in addition to events and messages, there would be a third tab that let's [...]

Definitely if we are getting stuff like OP25 does via GNUPlot would be a HUGE benefit. constellation and spectral display ie: functions 2, and 1 on OP25.

Tuning is best done to get the BEST AND TIGHTEST CONSTELLATION.. the rest after this will take care of its self 99.9999% of the time.

jpryor223, 2020-06-07:
control channel decode % or quality indication would be ideal to see in now playing, and the more detailed channel/tuner level view for diagnostic and trouble shooting when drilling deeper sounds ideal; at times i see i drop a control channel to idle, currently dont have a great way to see the quality if control channel at quick glance, or overall channels for more detailed tuning and adjustments