DSheirer/sdrtrunk #757: Broadcastify calls bounty

davidmckenzie, 2020-04-24:
https://www.rtl-sdr.com/broadcastify-calls-radio-call-archiving-incentives-for-sdr-software-developers/

> If developers adapt their SDR based application to function as a call ingest platform for Broadcastify Calls, Broadcastify will pay them up to $2,500 for their efforts. Here are the details:

Might be worth a look :)

ryanthomp, 2020-04-24:
It seems fairly simple given that SDRTrunk already records call segments and sends them in order to Broadcastify, seems like all that would be needed is to change where the data is sent and hook into the Broadcastify Calls API and send in the data that way.

DSheirer, 2020-04-24:
@davidmckenzie @ryanthomp I've been emailing with Lindsay about this.  He's setup an account for me to develop and test against.  It's in the work hopper.

DSheirer, 2020-04-26:
API: https://wiki.radioreference.com/index.php/Broadcastify-Calls-API

khanfar, 2020-05-01:
any useful link how to stream to broadcastify step by step guide ?

PR #773 (2020-05-02): #769 Broadcastify Calls API Streaming Feature
Resolves #757 #769

Implements Broadcastify Calls API audio broadcaster and resolves audio streaming error introduced with #5
closed 2020-05-02 by DSheirer