DSheirer/sdrtrunk #758: Disabling a traffic channel

mikerobertsn, 2020-04-25:
Hi, the traffic channels which automatically get created, when I disable this what does it do, does it just disable that call (and allow future calls) or does it permanently disable the caller etc.

DSheirer, 2020-04-25:
@mikerobertsn I only disables that traffic channel for that moment.  It doesn't do anything permanently.

You can post questions like this in the user email support group at https://groups.google.com/forum/#!forum/sdrtrunk

You can also post questions on the discord server at: https://discord.gg/HJQaKYE

Please limit using the github issues for reporting bugs or requesting features/enhancements.

closed 2020-04-25 by DSheirer

mikerobertsn, 2020-04-25:
Sure, will do next time, also thanks for the great program, it works really well!!