DSheirer/sdrtrunk #762: RR Is Currently Unavaliable

ryanthomp, 2020-04-26:
When I was testing out the RR interface in the fork it worked perfectly now that I downloaded from master I get this error in the playlist editor.

`13:19:06.935 ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [64MB/110MB 58%]`
[image]

ryanthomp, 2020-04-26:
Never mind took a couple of restarts and it finally connected

closed 2020-04-26 by ryanthomp

fofeResonance, 2020-07-22:
Hi!
I'm having the same issue described above. I've tried several times to no avail.

`12:40:13.754 ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [81MB/154MB 52%]`

bnmelvin, 2020-08-10:
Same issue here ... retries never seem to get to 100% ...

20200810 153257.352 [sdrtrunk pool-1-thread-8] ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [219MB/460MB 47%]
20200810 153303.156 [sdrtrunk pool-1-thread-7] ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [306MB/460MB 66%]
20200810 153311.513 [sdrtrunk pool-1-thread-6] ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [135MB/460MB 29%]
20200810 153315.530 [sdrtrunk pool-1-thread-4] ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [211MB/460MB 45%]
20200810 153403.057 [sdrtrunk pool-1-thread-7] ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [378MB/460MB 82%]
20200810 153406.896 [sdrtrunk pool-1-thread-3] ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [358MB/460MB 77%]
20200810 153411.854 [sdrtrunk pool-1-thread-3] ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [282MB/460MB 61%]
20200810 153415.648 [sdrtrunk pool-1-thread-6] ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [244MB/460MB 53%]
20200810 153419.172 [sdrtrunk pool-1-thread-1] ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [368MB/460MB 80%]
20200810 153422.641 [sdrtrunk pool-1-thread-8] ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [196MB/460MB 42%]

DSheirer, 2020-08-11:
@fofeResonance and @bnmelvin Can you both confirm that you have a current paid premium subscription to radio reference and that you're running the software on a machine that has good access to the internet?

awipslinuxuser, 2021-01-01:
Is there any free alternative to sdrtrunk that can decode p25 phase 1&2 besides openear