DSheirer/sdrtrunk #766: Frequency error and alias  0.5.0a1

ah1102, 2020-04-29:
I compiled the program from the source code using the latest version of JDK14. Created a new playlist and a new alias

1) The frequency is not displayed on the “Channels - Source” tab, and in the “Channels” there is no column with the site frequency.

[image]

2) When cloning an “Alias” with two or more identifiers, not one clone is created, but several.

[image]

ah1102, 2020-04-30:
Without these corrections, the program cannot be used. I don’t see the frequency of sites from my playlist. Moreover, the program can overwrite the frequencies stored in the playlist. I already had to restore the playlist. Editing a playlist is also difficult, because when I add a new identifier, I use the clone function. But this does not work correctly. At least for me.

DSheirer, 2020-05-03:
The frequency column is added.  I'm still working on the alias cloning issue.

PR #791 (2020-05-04): #766 Playlist Editor Clone Alias Causes Multiple Alias Copies
 Resolves #766 

Resolves issue with playlist alias editor clone alias feature causing multiple copies of alias.  Changed clone feature to only use a shallow copy (no ids or actions).  Removed alias event framework from code since all change notifications can now be detected by the alias model's aliasList property.
closed 2020-05-04 by DSheirer