DSheirer/sdrtrunk #798: Duplication of id when monitoring one site. 0.5.0a1

ah1102, 2020-05-06:
Duplicate identifier when monitoring a single site. Periodically, an echo occurs, since in two channels on the left and on the right, one identifier is received at the same frequency.

[image]

DSheirer, 2020-05-25:
The events table is showing exactly what the control channel transmits.  For non-call events like these, it would be difficult to track the duplication of events.

jpryor223, 2020-06-07:
since P25 outbound signaling (OSWs) control channel often contain the same message multiple times, this kind of duplication is seen often in messages and event log; this may be best to group like info together for onscreen display within the same time period/second or so (same event, from, to, details) to only show one unique entry on screen, and in the file based log, all details could be maintained if needed to troubleshoot further; from what I have seen, Unitrunker does a decent job of only showing the unique entries