DSheirer/sdrtrunk #801: Crash in the “Alias” and “CONTROL” color of the channel. 0.5.0a1

ah1102, 2020-05-06:
When you right-click in the "Now Playing" window and select the "Disable" menu, for example, one of the two sites, such a crash appears in "Allias" and the color "CONTROL" of the remaining site disappears.

[image]

DSheirer, 2020-05-07:
Note: there's a possibility that the context menu actions used here are invoking FX actions that are not on the FX thread.

DSheirer, 2020-05-09:
This was a difficult bug to track down.  When the playlist editor issues a stop channel to the channel processing manager, which in turn removes the channel metadata from the channel metadata model, the model fires a table row delete notification for row with index of -1.  This is apparently a sentinel to JTable to fully recreate all of the table columns from the model, which causes the JTable to lose all customer table cell renderers.  I added a check to ensure that we never issue a fireTableRowDeleted() for a table row with index -1.

PR #807 (2020-05-09): #797 and #801 Code Review and Cleanup.
Resolves #797 
Resolves #801 

Code Review and Cleanup.  Resolves issue with Swing JTable column updates causing custom cell renderers to disappear.
closed 2020-05-09 by DSheirer