DSheirer/sdrtrunk #806: Changing settings in playlist editor causes playing issues

ryanthomp, 2020-05-09:
When I change settings in the playlist editor this happens in the now playing section

[image]

DSheirer, 2020-05-09:
Ryan, do you remember which setting you were changing that caused this?

ryanthomp, 2020-05-09:
It happens alor for various settings but specifically when this happened I
played and stopped a passport channel.

DSheirer, 2020-05-09:
Perfect.  There's another user that's reported (I think) the same issue.  I'll get it fixed this morning.

ah1102, 2020-05-09:
Yes, the same problem. This happens when you disconnect and enable a channel from a playlist.

DSheirer, 2020-05-09:
Closing issue.  Duplicate to #801

closed 2020-05-09 by DSheirer