DSheirer/sdrtrunk #813: Playlist Editor 0.5.0a1

ah1102, 2020-05-10:
1. When creating a new channel and adding frequency to the table, the value 0 does not change.

2. Also there is no editing function “Protocol”. That is, I can not change the AM detector to NBFM without deleting the channel and creating a new one.

3. In the newly created playlist, the names “Alias” from another playlist are available, but without parameters. Is it intentional or a mistake?

[image]

DSheirer, 2020-06-10:
Since there are 4 separate issues being reported in this ticket, I've broken them up.
1. Addressed as issue #849

DSheirer, 2020-06-10:
Nick, I don't understand your issue #3 above.  When you add a colon (:) to the end of your channel name, what is happening or not happening?