DSheirer/sdrtrunk #814: P25 Phase 1 Channels Not Tearing Down

DSheirer, 2020-05-11:
Occassionally, there are P25 Phase 1 channels that get stuck in TEARDOWN state and do not get removed, forcing the user to either manually remove them (right click disable), or restarting the software to get the channel to disable.

bx-api, 2020-06-02:
I had this issue yesterday with 2 rtlsdr on MacBook Pro with usb3.  The lockup was happening quite frequently to me.  I noticed the last thing in the log was auto adjusting PPM.  I disabled that feature and then it didn’t lock up all night

jpryor223, 2020-06-07:
i have noticed this when interacting with the log and/or playlist editor, once i had accidentally clicked and held in the log, and had paused some activity, causing memory to spike and multiple channels left in teardown state; i have also seen some interactions and changes in playlist editor if saving while running, at times have left some channels in teardown state

closed 2026-11-06 by DSheirer