DSheirer/sdrtrunk #815: Airspy Mini Spectral Issues With 3.0 MHz Sample Rate

DSheirer, 2020-05-11:
User reported issue decoding P25 control channel when using airspy mini at 3.0 MHz sample rate.  Screen shot of spectral display reflected very wide filtering drop off across the right-most 25% of the spectral display.  The poor decodes were happening when the decode channel frequency landed in this region.

Investigate if this is an issue with the airspy mini, or if sdrtrunk is not configuring something correctly fo  r the 3.0 MHz sample rate setting on the mini.

aaomidi, 2020-06-04:
Going to be getting an airspy mini, so let me know if i can help with any investigations

DSheirer, 2020-09-28:
Change airspy tuner controller to use the following sample rates and usable spectrum settings:
10 MHz - 90%
2.5 MHz - 60%
6 MHz - 83%
3 MHz - 66%

PR #976 (2020-10-25): #815 - Change usable bandwidth percentage based on Airspy samplerate
Percentages based on @DSheirer's percentages given in the bug report.

Tested with the Airspy Mini, and the percentages seem to be good. The usable bandwidth with Airspy Mini at 3MHz is asymmetric, but probably close enough to be fine as it is.

Fixes #815 .
commit c6afb1c1 (2020-10-25): Fixes #815 - Change usable bandwidth percentage based on Airspy samplerate (#976)
closed 2020-10-25 by DSheirer