DSheirer/sdrtrunk #817: Jittery traffic channel audio for MPT1327

apankowski, 2020-05-14:
Audio seems to drift between choppy/jittery and back to normal for MPT1327 on my setup, which is:

* Windows 10 x64,
* JDK 14,
* sdrtrunk 0.4.0 as well as 0.5.0-alpha1 (directly from `master` branch),
* R820T as well as Airspy R2.

I'm attaching samples of both the drifting traffic channel audio as well just the jittery/choppy part: 
[jittery-voice-samples.zip](https://github.com/DSheirer/sdrtrunk/files/4630866/jittery-voice-samples.zip)

What I've tried:

* different versions of sdrtrunk mentioned above,
* different tuners mentioned above,
* changing the channelizer from polyphase to heterodyne,
* changing tuner sampling rate from 0.3 all the way to 10 mbps,
* decoding different MPT systems,

but nothing of the above helped.

I'm not sure on how would I go about finding the root cause of this issue, especially since I don't know the app and its inner parts that well yet. Are there any classes/packages for which I could turn logging on to debug the processing chain? I will be happy to provide more details on my end as well as samples of baseband for the traffic channel, etc.

DSheirer, 2020-05-15:
@apankowski It sounds like it's duplicating each and every audio chunk that get's assembled into the audio segment.

I'll have to dig into this more, because nothing jumps out at me that could be causing this.  The only way that I can think that would cause the audio to duplicate like that would be if there were 2 AudioModules added to the processing chain for the traffic channel instead of just one.  But, each time a processing chain is reused, all of the modules are removed and then replaced with a fresh set of modules.  The DecoderFactory controls which modules get added for each decoder type and it specifies just one audio module for MPT1327.

DSheirer, 2020-05-15:
Looking at it some more, the audio is not completely duplicated, so that seems to eliminate the duplicate AudioModule theory.  

I'm thinking this issue might be in the audio resampler.  I'll have to setup some testing to see if I can reproduce it.

apankowski, 2020-05-22:
Update:

The samples I've attached before were from Windows audio loopback ("record what you hear").

Trying to find the cause of this issue - and using the same setup - I enabled recording of conversations in `.wav` format via "aliases". The result is peculiar: for a single conversation I get couple of `.wav` files. Every file has 8kHz sample rate consisting of 32-bit float samples (both expected). But every file except for the last one has exactly 125 000 samples and it seems to be a 1000-sample loop repeated 125 times. Sample here:

[TRAFFIC__TO_825500_FROM_819705.zip](https://github.com/DSheirer/sdrtrunk/files/4667545/TRAFFIC__TO_825500_FROM_819705.zip)

Every file has different loop samples, but all seem to have at most 125 000 samples (last file is an exception here) and consist of 1000-sample loop repeated 125 times.

I am unsure if this is related to the reported issue or if it's a separate one - I will try to investigate.

Now, after enabling traffic channel baseband recording I obtained a longer `.wav` file. I was able to open and process it with SDR# and I heard the conversation without any jitter/choppiness. So it seems that traffic channel baseband streaming is correct, but the jitter/choppiness is introduced at a later processing stage.

I could use your wisdom here @DSheirer - can you point me to the top-level classes implementing later stages of processing pipeline (including the audio recording class)?

DSheirer, 2020-05-22:
@apankowski I'll try to dig into this tonight after work, or maybe tomorrow morning.  At the moment I can't think of anything to point you to for debugging.  

The only thing that comes to mind is you might watch the Squelch State monitoring in the Audio Module to see if the SquelchStateMonitor is triggering and causing multiple audio segments.  The squelch state is controlled by the SingleChannelState as the state machine changes states with inputs from the MPT1327DecoderState.

DSheirer, 2020-05-24:
I figured it out.  The AudioModule was not creating a copy of the audio samples from the high-pass filtering operation.  It was using a reference to the audio sample array that was being overwritten by each subsequent audio buffer that came in.

closed 2020-05-24 by DSheirer

apankowski, 2020-05-25:
@DSheirer just wanted to confirm that on my end both the jitter in traffic channel audio and these weird loops in recordings are gone after your fix 🎉

Thank you 🙂

PR #830 (2020-05-24): #817 MPT1327 Jittery Audio Playback and Recording
Resolves #817

Resovles issue with jittery audio playback and recording for MPT1327 channels.