DSheirer/sdrtrunk #820: MPT1327 - Call timeout is effectively ignored

apankowski, 2020-05-17:
# The problem

It seems that call timeout setting of MPT1327 decoder is effectively ignored.

In my setup I've set the timeout to 120 seconds:

[image]

However, for calls longer than 45 seconds, traffic channel decoding stops after about 45 seconds:

[image]

Part of the spectrum marked by red rectangle shows the traffic channel - as you can see there's a conversation still going on.

After some debugging it seems this behavior is caused by this line in `SingleChannelState`:

https://github.com/DSheirer/sdrtrunk/blob/3b357c4d40cdc990b829d63d59d1528c45465cf3/src/main/java/io/github/dsheirer/channel/state/SingleChannelState.java#L118

... which seems to stop forwarding audio for any traffic channel after `DEFAULT_CALL_TIMEOUT_DELAY_SECONDS` (45) seconds.

# Testing this theory

After replacing usage of mentioned constant with `120_000` (120 seconds) I was able to follow a conversation longer than 120 seconds:

[image]

Part of the spectrum marked by red rectangle shows the traffic channel - again, the conversation is still going on: both parts of the call are silent at the moment, but `CLEAR` message wasn't sent just yet. Traffic channel decoding stopped after hardcoded timeout.

# Possible solutions

It seems that call timeout setting for MPT should be used in mentioned line. Couple of fixes come to mind:

## 1. Quick but ugly

Just apply the call timeout for MPT in `SingleChannelState`. More or less:

```java
final int fadeTimeoutSeconds;
if (channel.getDecodeConfiguration() instanceof DecodeConfigMPT1327) {
    fadeTimeoutSeconds = ((DecodeConfigMPT1327) channel.getDecodeConfiguration()).getCallTimeout();
} else {
    fadeTimeoutSeconds = DecodeConfiguration.DEFAULT_CALL_TIMEOUT_DELAY_SECONDS;
}
mStateMachine.setFadeTimeoutBuffer(fadeTimeoutSeconds * 1000);
```

## 2. Introduce a trait interface

```java
interface WithCallTimeout {
    int getCallTimeout();
}
```

Make any interested decoder config implement it:

```java
class DecodeConfigMPT1327 extends DecodeConfiguration implements WithCallTimeout {
}
```

Apply call timeout in `SingleChannelState` based on the trait:

```java
final int fadeTimeoutSeconds;
if (channel.getDecodeConfiguration() instanceof WithCallTimeout) {
    fadeTimeoutSeconds = ((WithCallTimeout) channel.getDecodeConfiguration()).getCallTimeout();
} else {
    fadeTimeoutSeconds = DecodeConfiguration.DEFAULT_CALL_TIMEOUT_DELAY_SECONDS;
}
mStateMachine.setFadeTimeoutBuffer(fadeTimeoutSeconds * 1000);
```

## 3. Add call timeout for all protocol decoders

I.e. to `DecoderConfiguration` base class:

```java
public abstract class DecodeConfiguration extends Configuration
{
    @JsonIgnore
    public int getCallTimeout()
    {
        return DEFAULT_CALL_TIMEOUT_DELAY_SECONDS;
    }
}
```

Then override as needed for specific decoder, e.g. MPT:

```java
@Override
@JsonIgnore(false) // A bit ugly, but needed
@JacksonXmlProperty(isAttribute = true, localName = "call_timeout")
public int getCallTimeout()
{
    return mCallTimeout;
}
```

And of course apply in `SingleChannelState`:

```java
mStateMachine.setFadeTimeoutBuffer(channel.getDecodeConfiguration().getCallTimeout() * 1000);
```

Although I'm not sure if saying that every decoder has a "call timeout" is semantically correct as I don't know much about e.g. P25 or Passport.

***

Please let me know if you think one of these solutions is the way and I will happily submit a PR :slightly_smiling_face:

DSheirer, 2020-05-20:
@apankowski Sorry it took so long to respond on this.  I like option 2.  

One additonal aspect to this is the AbstractAudioModule imposes a max audio segment length limit of 60 seconds.  Right now this is a constant, but a member variable could be added with public method that could be set int the DecoderFactory when it constructs the primary modules for the MPT traffic channel.

apankowski, 2020-05-20:
@DSheirer Nah, I think you're replying quickly 🙂

Cool, I like option 2 as well. I will get to it.

Regarding `AbstractAudioModule` and max audio segment length... After reading through the code a bit I can't see how this could affect max length of audio playback. If I understand correctly the audio from e.g. a traffic channel doesn't have to fit in one segment, no? And so max audio segment length is just an upper limit on the _segment_ length, but after reaching it a new one will be created and linked with the previous one:

https://github.com/DSheirer/sdrtrunk/blob/8167e9e9f2c07480a45fd417653e419b4ae016a0/src/main/java/io/github/dsheirer/audio/AbstractAudioModule.java#L120

... so that the playback manager knows which output should it be passed to (in other words: to support continuity of the audio stream):

https://github.com/DSheirer/sdrtrunk/blob/8167e9e9f2c07480a45fd417653e419b4ae016a0/src/main/java/io/github/dsheirer/audio/playback/AudioPlaybackManager.java#L186

If the above is true it would mean that `MAX_SEGMENT_AUDIO_SAMPLE_LENGTH` is an arbitrary constant and it could just as well be `8000 * 5` (5 seconds as 8 kHz) in which case the segments would just be swapped (reserved and released) more often.

Am I missing something here?

DSheirer, 2020-05-20:
No, you're correct.  The only thing that audio segment length will impact is recording.  If you set the traffic channel timeout to 2 minutes, this would generate 2 recordings for the same audio call.  The Recording Manager will create a recording for each audio segment.

PR #823 (2020-05-21): MPT1327: Fix for call timeout setting not being honored
Fixes #820 

Any requests, remarks & pointers highly welcome 🙂
closed 2020-05-21 by DSheirer