DSheirer/sdrtrunk #828: Add Support for Zello Streaming

DSheirer, 2020-05-22:
https://github.com/zelloptt/zello-channel-api/blob/master/API.md

bx-api, 2020-05-31:
this would be amazing!  

I currently have 2 laptops and 4 dongles to be able to get police and fire on separate zello channels since I'm using virtual audio cable.  It would sweet to reduce my setup to 1 laptop.

LordDonkey, 2020-06-07:
I concur! This would be an excellent addition to SDRTrunk. With this, I could eliminate the VAC and Zello all in one go, it would definitely give back much needed system resources.

josephdambrosio, 2020-12-03:
+1 for this

mikebuckley, 2021-03-14:
+1 for this, please!

Deziel0495, 2025-06-26:
+10 for this.

Sending audio via VOX works, but isn't ideal. Having official support for Zello within SDRTrunk would be a game changer!

Deziel0495, 2026-10-13:
@DSheirer is this still on the roadmap?