DSheirer/sdrtrunk #836: Invalid XML in playlist

hunterbeach, 2020-05-28:
Recently I was editing channels in the playlist editor when it asked multiple times to save the alias configuration, to which I hit yes. Shortly after the program crashed and the playlist was gone. I loaded a backup before it crashed and it said it was an invalid playlist. Looking at the error, I could see the two lines it was tripped up on and they were the two aliases being edited right before the crash. Here are the two problem aliases:

`<alias name="Tulsa Transit B" color="-16776961" group="Public Works" list="OKWIN" iconName="Garbage Truck">
    <id type="priority" priority="-1"/>
    <id/>
  </alias>
  <alias name="Tulsa Transit C" color="-16776961" group="Public Works" list="OKWIN" iconName="Garbage Truck">
    <id type="priority" priority="-1"/>
    <id/>
  </alias>`

As you can see there are two `<id/>` tags that were the cause of the problems. I removed those two tags and the playlist loaded normally. This is on v0.5.0 alpha1

NoahWL, 2020-07-06:
On v0.50 alpha2, just had this happen while editing a single alias that I had found using the search bar.  I was asked multiple times to save the alias configuration despite having already pressed "save."  Clicking yes just re-opened the prompt.  Clicking no closed the prompt but the change was not saved, and the console filled with an error.  It seems the error repeats depending on how many aliases entries I change before the dialog displays.

The alias I was attempting to edit at the time now has three "priority" attributes in the XML file:
```
<alias name="DwnrsGrove Local" color="-16711681" group="DuPage County - Local" list="STARCOM21 LO">
    <id type="priority" priority="4"/>
    <id type="priority" priority="4"/>
    <id type="priority" priority="4"/>
    <id type="talkgroup" value="2692" protocol="APCO25"/>
  </alias>
```

For a bit I was able to reproduce the dialog pop-up and the error but I seem to have forgotten how...  I'm going to keep looking at how this may have occurred and I'll update if I find anything noteworthy.

e850205, 2020-07-06:
@NoahWL I have the same problem about 25-50% of the time when I try to save a new alias. I hope you can figure it out because it's a very annoying bug!

NoahWL, 2020-07-09:
Figured it out and have a fix.  I realize now it's a bit different than the issue original posted here.  When I have time this week I'll open a new issue with more details and a pull request for the fix.