DSheirer/sdrtrunk #841: No Audio on Any OS

sutterlaird, 2020-06-03:
Hello all,

I know I must be doing something wrong but I just can't figure out what it is. I have installed sdrtrunk and JMBE on my MacBook Pro running macOS Mojave, an Ubuntu 20.04 VM in VirtualBox, and a Windows 10 VM in Parallels Desktop but I have not been able to get audio to work on any OS. I have the control channel set up, I see Messages and Events coming, and traffic channels are being created, but no sound is generated and the Audio Panel just shows "------" for both channels. Terminal confirms that JMBE is being loaded correctly. I've tried sdrtrunk 0.5 and 0.4 and JMBE 1.0.6 and 1.0.7 on all three platforms. I've attached screenshots of my configuration, do you see what I am missing?

Thank you,
Sutter

[image]
[image]
[image]
[image]

DSheirer, 2020-06-03:
It looks like you're using the wrong control channel frequency.  Radio reference has it listed as 770.34375.

Change the channel configuration frequency from 770.25 to 770.34375 and then adjust the tuner PPM so that the control channel signal that you're currently decoding aligns with the frequency readout of 770.34375.

In your screen shot, you can see where sdrtrunk thinks the traffic channel should be, and then you can see the actual signal that sdrtrunk should be decoding is about 9.375 kHz off.  Once you adjust the tuner PPM, everything should line up and decode properly.

DSheirer, 2020-06-03:
When a traffic channel is allocated, click on the traffic channel row in the now playing window and make sure you're seeing decoded messages.  If you were to click on the traffic channel as it is configured right now, you're probably seeing nothing but sync loss messages, indicating that the traffic channel is not decoding correctly.

sutterlaird, 2020-06-03:
Thank you very much, that was what I was missing. I am now hearing some traffic. I dialed in the PPM to get the frequency as close as I could but I'm still seeing a lot of sync loss messages. My SDR dongle is an early generic model, would I have better luck with a better unit, like an authentic RTL-SDR unit?

DSheirer, 2020-06-03:
The earlier screenshots look like your gain may be too high.  The generic dongles work ok once you get them adjusted

sutterlaird, 2020-06-03:
I'll try experimenting with the gain. Thank you again!

closed 2020-06-03 by sutterlaird