DSheirer/sdrtrunk #842: Out Of Memory Errors after 12-16 hours

bfranske, 2020-06-05:
I'm getting "java.lang.OutOfMemoryError: Java heap space" errors in the console after 12-16 hours of monitoring and the program locks up and needs to be restarted. I'm running 0.5.0-alpha1 on a very large P25 phase I system (over 4000 TGs/aliases) using two RTL-SDR dongles.

Thanks!
-Ben

LordDonkey, 2020-06-05:
Also experienced memory errors with the latest build a1e8c76. It ran for 8 hours 34 mins before locking up and issuing the same error. It jumped from 925 MB of RAM to 3GB in less than 2 minutes. Memory issues weren't present with release version 89cb546.

`20200603 184948.035 [sdrtrunk pool-1-thread-5] INFO  i.g.d.module.log.EventLogger - Creating log file:C:\Users\ICBus\SDRTrunk\event_logs\20200603_184948.035_774031250_Hz_TRAFFIC_decoded_messages.log  [925MB/1GB 84%]
20200603 185132.434 [sdrtrunk pool-1-thread-4] INFO  i.g.d.module.log.EventLogger - Creating log file:C:\Users\ICBus\SDRTrunk\event_logs\20200603_185132.434_774031250_Hz_TRAFFIC_call_events.log  [3GB/3GB 99%]
20200603 185313.402 [NioProcessor-24] INFO  i.g.d.a.b.AudioStreamingBroadcaster - [GCFirst] status: Disconnected  [3GB/3GB 99%]
20200603 190419.033 [sdrtrunk pool-1-thread-2] ERROR i.g.d.s.t.u.USBTransferProcessor - [R820T 00000001] - error while processing USB transfer buffers  [3GB/3GB 99%]
java.lang.OutOfMemoryError: Java heap space
20200603 190410.708 [sdrtrunk pool-1-thread-4] INFO  i.g.d.module.log.EventLogger - Creating log file:C:\Users\ICBus\SDRTrunk\event_logs\20200603_190404.818_774031250_Hz_TRAFFIC_decoded_messages.log  [3GB/3GB 99%]`

bfranske, 2020-06-19:
So after alpha2 runs for some indeterminate amount of time I see my logging screen filled up with:
`10:10:13.127 ERROR i.g.d.a.c.MP3AudioConverter - There was an error converting audio to MP3: null  [1GB/1GB 99%]
10:10:13.127 ERROR i.g.d.a.c.MP3AudioConverter - There was an error converting audio to MP3: null  [1GB/1GB 99%]
10:10:14.066 ERROR i.g.d.a.c.MP3AudioConverter - There was an error converting audio to MP3: null  [1GB/1GB 99%]
10:10:14.066 ERROR i.g.d.a.c.MP3AudioConverter - There was an error converting audio to MP3: null  [1GB/1GB 99%]
10:10:14.923 ERROR i.g.d.a.c.MP3AudioConverter - There was an error converting audio to MP3: null  [1GB/1GB 99%]
10:10:14.955 ERROR i.g.d.a.c.MP3AudioConverter - There was an error converting audio to MP3: null  [1GB/1GB 99%]
10:10:15.923 ERROR i.g.d.a.c.MP3AudioConverter - There was an error converting audio to MP3: null  [1GB/1GB 99%]
10:10:15.954 ERROR i.g.d.a.c.MP3AudioConverter - There was an error converting audio to MP3: null  [1GB/1GB 99%]`
And the control channel shows as idle with no calls going on which is certainly not true. As soon as I shutdown and restart SDRTrunk things come back.

LoveScanner, 2022-08-16:
Did you find a solution?

bfranske, 2022-08-16:
No, I did write script that would kill and restart sdrtrunk every 6 hours as a stopgap. Ultimately I moved my "production" use of sdrtrunk to trunk-recorder on Linux instead and just use sdrtrunk for short term testing.

closed 2023-11-10 by DSheirer