DSheirer/sdrtrunk #848: BladeRF / Micro

gusosborne, 2020-06-10:
Haven’t been able to find any reference to blade here or on the groups - would be killer to utilize the ~60Mhz bandwidth though. 
Any chance for BladeRF (Micro) support? Possibly via soapy?

DSheirer, 2020-06-12:
@gusosborne I don't have a bladeRF, so I probably won't be able to add support for this.

gusosborne, 2020-06-12:
Yea, it would probably only make sense via soapy. You don’t have any plans for adding soapy support then?
Either way - thank you for this awesome tool. This finally got my father doing P25 trunked traffic.

l0g-lab, 2024-12-16:
I have began working on this on my fork. I'm no java developer by any stretch so no promises on it being completed, but I have a bladerf and would like to see it work.

I'm basing a lot of it on the way sdrtrunk works with the hackrf, just migrating to using libbladerf's API vs libhackrf. We'll see how it goes...

l0g-lab, 2024-12-27:
To build support for the existing SDRs, did you ( @DSheirer ) reverse the low level USB control codes etc to make the SDR work? I don't see importing any libraries for anything. I'm using hackrf as a boilerplate for adding support for bladerf and I found the enumerations that you included, but bladerf is built much different seemingly and reversing it is... fun lol.

DSheirer, 2026-11-06:
Closing. jBladerf library now integrated

closed 2026-11-06 by DSheirer