DSheirer/sdrtrunk #859: Display of frequency in input line “Source” 0.5.0

ah1102, 2020-06-11:
When assembling the program from source codes with the latest fixes, a problem was found with displaying the frequency in the “Source” input line. Maybe something is broken? I use to build JDK 14.0.1
[image]

In the author's release - "0.5.0 alpha 1" there is no such problem.
[image]

DSheirer, 2020-06-12:
Nick,
I just updated the gradle build script to use a newer (13.0.2) set of JavaFX libraries.  Can you do a git pull and build and see if that clears the font rendering issues you're seeing?

ah1102, 2020-06-12:
Yes

ah1102, 2020-06-12:
It seems to me that he has become a little better or remained unchanged.

[image]

DSheirer, 2020-06-12:
When you click the 456.0 MHz line in the picture above, is the frequency value loading into the Source editor?

ah1102, 2020-06-12:
No, in the program that I compiled from this code - https://github.com/DSheirer/sdrtrunk/releases/tag/v0.5.0-alpha.1

In your assembly sdr-trunk-osx-x86_64-v0.5.0-alpha1.zip everything is fine!

DSheirer, 2020-06-12:
Can you build from the master branch and test tha? 

 (not the v0.5.0-alpha.1 tag)

ah1102, 2020-06-12:
Yes, I did "git pull", as you requested from the main branch https://github.com/DSheirer/sdrtrunk.git and posted you a screenshot where the font is not visible to change.

Also, there is no frequency display in the "Source" line:

`When you click the 456.0 MHz line in the picture above, is the frequency value loading into the Source editor?`

DSheirer, 2020-06-12:
Can you post the playlist file that you're using for this test.  I want to see if there is anything odd about the channel where the frequency value is not displaying in the source frequency field.

ah1102, 2020-06-12:
Ok

[test.xml.zip](https://github.com/DSheirer/sdrtrunk/files/4770160/test.xml.zip)

DSheirer, 2020-06-12:
The playlist looks good.  I have to go to work ... I'll try to look into this more over the weekend.

ah1102, 2020-06-12:
Ok, thanks for the help! Have a nice day!!

ah1102, 2020-06-12:
I found that in the program that I compiled from the source code, fewer files:
[image]

Here is your folder and there are more files in it:
[image]

DSheirer, 2020-06-12:
The difference is you're building and using your local JDK14.  

The release version has its own (internal) JDK 14 version.

ah1102, 2020-06-12:
Maybe I'm building the program incorrectly? I recorded a video of the process https://youtu.be/rmfWq9lShcc

ah1102, 2020-06-13:
I downloaded your release "Version 0.5.0 Alpha 2" and everything is fine in it, except for rendering fonts. The frequency in the line is displayed. What could be wrong with the assembly on my computer? Maybe somewhere you need to run the sudo command?

ah1102, 2020-06-18:
I changed the localization of the system interface from Russian to English as the main one. And now, the numbers have appeared! Surprisingly, when you return Russian as your main language, they are no longer displayed.
I think that with other regional localizations this may also be the case.

ah1102, 2020-06-18:
For example, I tried to install Finnish as the macOS system language, and the numbers stopped showing.

Suomi
[image]

English
[image]

ah1102, 2020-09-09:
Danny, are you registered with the Java bug tracker? Can you create a ticket?