DSheirer/sdrtrunk #866: Cellocator Packet Parsing

DSheirer, 2020-06-13:
Add support for parsing Cellocator Telemetry over APCO25 data packets.

ryanthomp, 2020-06-13:
Would this go across in data calls such as the ones I sent you?

DSheirer, 2020-06-13:
Possibly.  I just discovered this today.  I was getting a bunch of unknown UDP packets on the local Phase 1 system.  All of the packets are going to UDP port 231 and they start with "4D434750" which is ASCII 'MCGP' and is the header sentinel value for Cellocator messages

ryanthomp, 2020-06-13:
Very interesting, once you commit this I will download the master branch and let you know if I get anything.

DSheirer, 2020-06-15:
Some interim output ...
```
07:44:07.292  LLID:894282 TO:10.172.193.174 UDP PORT FROM:231 TO:231 CELLOCATOR ACKNOWLEDGE UNIT:568929 LOCATION AND STATUS MESSAGE NUMBER:175 TIME:Sat Jun 13 10:31:12 EDT 2020 AUTH:DDD80ACF  [42MB/252MB 16%]
07:44:07.379  LLID:849208 TO:10.172.192.243 UDP PORT FROM:231 TO:231 CELLOCATOR PROGRAM RADIO:564534 SEQUENCE NUMBER:14 AUTH:0 BLOCK:6 BITMASK:0004 DATA:00000000000000000000850000000000  [8MB/252MB 3%]
07:44:07.388  LLID:849208 TO:10.172.192.243 UDP PORT FROM:231 TO:231 CELLOCATOR PROGRAM RADIO:564534 SEQUENCE NUMBER:15 AUTH:0 BLOCK:6 BITMASK:0800 DATA:00000000000000000000000000000000  [15MB/252MB 6%]
07:44:07.418  LLID:894282 TO:10.172.193.174 UDP PORT FROM:231 TO:231 CELLOCATOR ACKNOWLEDGE UNIT:568929 LOCATION AND STATUS MESSAGE NUMBER:176 TIME:Sat Jun 13 10:31:33 EDT 2020 AUTH:DDD80ACF  [23MB/252MB 9%]
07:44:07.430  LLID:894282 TO:10.172.193.174 UDP PORT FROM:231 TO:231 CELLOCATOR ACKNOWLEDGE UNIT:568929 LOCATION AND STATUS MESSAGE NUMBER:177 TIME:Sat Jun 13 10:31:33 EDT 2020 AUTH:DDD80ACF  [29MB/252MB 11%]
07:44:07.501  LLID:894187 TO:10.172.193.232 UDP PORT FROM:231 TO:231 CELLOCATOR ACKNOWLEDGE UNIT:569053 LOCATION AND STATUS MESSAGE NUMBER:120 TIME:Sat Jun 13 10:31:49 EDT 2020 AUTH:DDD80A55  [56MB/252MB 22%]
07:44:07.524  LLID:849208 TO:10.172.192.243 UDP PORT FROM:231 TO:231 CELLOCATOR RADIO:564534 REQUEST:IMMEDIATE STATUS AND LOCATION SEQUENCE:16 AUTH:0  [66MB/252MB 26%]
07:44:07.535  LLID:849208 TO:10.172.192.243 UDP PORT FROM:231 TO:231 CELLOCATOR PROGRAM RADIO:564534 SEQUENCE NUMBER:17 AUTH:0 BLOCK:6 BITMASK:0004 DATA:00000000000000000000850000000000  [73MB/252MB 29%]
07:44:07.554  LLID:849208 TO:10.172.192.243 UDP PORT FROM:231 TO:231 CELLOCATOR PROGRAM RADIO:564534 SEQUENCE NUMBER:17 AUTH:0 BLOCK:6 BITMASK:0004 DATA:00000000000000000000850000000000  [14MB/252MB 5%]
07:44:07.562  LLID:849208 TO:10.172.192.243 UDP PORT FROM:231 TO:231 CELLOCATOR PROGRAM RADIO:564534 SEQUENCE NUMBER:18 AUTH:0 BLOCK:6 BITMASK:0800 DATA:00000000000000000000000000000000  [21MB/252MB 8%]
07:44:07.582  LLID:849208 TO:10.172.192.243 UDP PORT FROM:231 TO:231 CELLOCATOR PROGRAM RADIO:564534 SEQUENCE NUMBER:18 AUTH:0 BLOCK:6 BITMASK:0800 DATA:00000000000000000000000000000000  [32MB/252MB 12%]
07:44:07.625  LLID:849208 TO:10.172.192.243 UDP PORT FROM:231 TO:231 CELLOCATOR RADIO:564534 REQUEST:IMMEDIATE STATUS AND LOCATION SEQUENCE:19 AUTH:0  [51MB/252MB 20%]
07:44:07.633  LLID:849208 TO:10.172.192.243 UDP PORT FROM:231 TO:231 CELLOCATOR PROGRAM RADIO:564534 SEQUENCE NUMBER:20 AUTH:0 BLOCK:6 BITMASK:0004 DATA:00000000000000000000850000000000  [57MB/252MB 22%]
07:44:07.646  LLID:849208 TO:10.172.192.243 UDP PORT FROM:231 TO:231 CELLOCATOR PROGRAM RADIO:564534 SEQUENCE NUMBER:20 AUTH:0 BLOCK:6 BITMASK:0004 DATA:00000000000000000000850000000000  [65MB/252MB 26%]
07:44:07.654  LLID:849208 TO:10.172.192.243 UDP PORT FROM:231 TO:231 CELLOCATOR PROGRAM RADIO:564534 SEQUENCE NUMBER:21 AUTH:0 BLOCK:6 BITMASK:0800 DATA:00000000000000000000000000000000  [72MB/252MB 28%]
07:44:07.666  LLID:849208 TO:10.172.192.243 UDP PORT FROM:231 TO:231 CELLOCATOR PROGRAM RADIO:564534 SEQUENCE NUMBER:21 AUTH:0 BLOCK:6 BITMASK:0800 DATA:00000000000000000000000000000000  [79MB/252MB 31%]
07:44:07.697  LLID:849208 TO:10.172.192.243 UDP PORT FROM:231 TO:231 CELLOCATOR RADIO:564534 REQUEST:IMMEDIATE STATUS AND LOCATION SEQUENCE:22 AUTH:0  [89MB/252MB 35%]
07:44:07.707  LLID:849208 TO:10.172.192.243 UDP PORT FROM:231 TO:231 CELLOCATOR PROGRAM RADIO:564534 SEQUENCE NUMBER:23 AUTH:0 BLOCK:6 BITMASK:0004 DATA:00000000000000000000850000000000  [95MB/252MB 37%]
07:44:08.321  LLID:894187 TO:10.172.193.232 UDP PORT FROM:231 TO:231 CELLOCATOR ACKNOWLEDGE UNIT:569053 LOCATION AND STATUS MESSAGE NUMBER:118 TIME:Sat Jun 13 10:25:49 EDT 2020 AUTH:DDD80A55  [88MB/252MB 35%]
07:44:08.385  LLID:894282 TO:10.172.193.174 UDP PORT FROM:231 TO:231 CELLOCATOR ACKNOWLEDGE UNIT:568929 LOCATION AND STATUS MESSAGE NUMBER:165 TIME:Sat Jun 13 10:26:12 EDT 2020 AUTH:DDD80ACF  [120MB/252MB 47%]
07:44:08.445  LLID:894282 TO:10.172.193.174 UDP PORT FROM:231 TO:231 CELLOCATOR ACKNOWLEDGE UNIT:568929 LOCATION AND STATUS MESSAGE NUMBER:166 TIME:Sat Jun 13 10:26:32 EDT 2020 AUTH:DDD80ACF  [145MB/252MB 57%]
07:44:08.456  LLID:893070 TO:10.172.194.222 UDP PORT FROM:231 TO:231 CELLOCATOR PROGRAM RADIO:4985105 SEQUENCE NUMBER:1 AUTH:0 BLOCK:6 BITMASK:0004 DATA:00B43800000000075A1C2D0000000000  [151MB/252MB 60%]
07:44:08.884  LLID:894282 TO:10.172.193.174 UDP PORT FROM:231 TO:231 CELLOCATOR ACKNOWLEDGE UNIT:568929 LOCATION AND STATUS MESSAGE NUMBER:168 TIME:Sat Jun 13 10:28:12 EDT 2020 AUTH:DDD80ACF  [95MB/303MB 31%]
07:44:08.923  LLID:894282 TO:10.172.193.174 UDP PORT FROM:231 TO:231 CELLOCATOR ACKNOWLEDGE UNIT:568929 LOCATION AND STATUS MESSAGE NUMBER:168 TIME:Sat Jun 13 10:28:12 EDT 2020 AUTH:DDD80ACF  [108MB/303MB 35%]
07:44:08.965  LLID:894282 TO:10.172.193.174 UDP PORT FROM:231 TO:231 CELLOCATOR ACKNOWLEDGE UNIT:568929 LOCATION AND STATUS MESSAGE NUMBER:169 TIME:Sat Jun 13 10:28:28 EDT 2020 AUTH:DDD80ACF  [128MB/303MB 42%]
07:44:08.980  LLID:894282 TO:10.172.193.174 UDP PORT FROM:231 TO:231 CELLOCATOR ACKNOWLEDGE UNIT:568929 LOCATION AND STATUS MESSAGE NUMBER:170 TIME:Sat Jun 13 10:28:33 EDT 2020 AUTH:DDD80ACF  [137MB/303MB 45%]
07:44:09.565  LLID:849208 TO:10.172.192.243 UDP PORT FROM:231 TO:231 CELLOCATOR PROGRAM RADIO:564534 SEQUENCE NUMBER:28 AUTH:0 BLOCK:6 BITMASK:0004 DATA:00000000000000000000850000000000  [105MB/303MB 34%]
[... cut ...]
07:44:10.580  LLID:893070 TO:10.172.194.222 UDP PORT FROM:231 TO:231 CELLOCATOR PROGRAM RADIO:568849 SEQUENCE NUMBER:242 AUTH:0 BLOCK:6 BITMASK:0800 DATA:00000002000000000000000000000000  [40MB/303MB 13%]
07:44:10.639  LLID:894185 TO:10.172.193.185 UDP PORT FROM:231 TO:231 CELLOCATOR ACKNOWLEDGE UNIT:569045 LOCATION AND STATUS MESSAGE NUMBER:188 TIME:Sat Jun 13 10:24:26 EDT 2020 AUTH:DDD80A51  [68MB/303MB 22%]
07:44:10.646  LLID:894282 TO:10.172.193.174 UDP PORT FROM:231 TO:231 CELLOCATOR ACKNOWLEDGE UNIT:568929 LOCATION AND STATUS MESSAGE NUMBER:160 TIME:Sat Jun 13 10:24:27 EDT 2020 AUTH:DDD80ACF  [74MB/303MB 24%]

```

ryanthomp, 2020-06-19:
Not sure if this means anything but is there any way to decode Motorola ARS packets? I also see a UDP Packet transfer between 4001 and 4002 with intermittent Motorola ARS packet transfers.

DSheirer, 2020-06-19:
The Motorola ARS packets are being decoded and those should be for port 4005.

The port 4001 packets should be for LRRP (GPS) position reporting and those currently are not being decoded.

ryanthomp, 2020-06-21:
What about port 2002 I see that being used often but no standard protocol just a bunch of Motorola deregister and registers and "ADD PATCH VOICE", but it is a call between 2 IPs at port 2002.

commit 7962758e (2020-07-25): #866 Cellocator Packet Support (#875)

Resolves #866 Implements cellocator packet support.
closed 2020-07-25 by DSheirer