DSheirer/sdrtrunk #867: Memory leak in beta

kenrestivo, 2020-06-14:
```
20200612 100858.771 [sdrtrunk pool-1-thread-4] INFO  i.g.d.module.log.EventLogger - Creating log file:/home/sdrtrunk/SDRTrunk/event_logs/20200612_100858.771_772693750_Hz_TRAFFIC_decoded_messages.log  [3GB/4GB 96%]
20200612 100901.373 [sdrtrunk pool-1-thread-7] INFO  i.g.d.module.log.EventLogger - Creating log file:/home/sdrtrunk/SDRTrunk/event_logs/20200612_100901.373_769768750_Hz_TRAFFIC_call_events.log  [3GB/4GB 96%]
20200612 100901.374 [sdrtrunk pool-1-thread-7] INFO  i.g.d.module.log.EventLogger - Creating log file:/home/sdrtrunk/SDRTrunk/event_logs/20200612_100901.374_769768750_Hz_TRAFFIC_decoded_messages.log  [3GB/4GB 96%]
20200612 100901.463 [sdrtrunk pool-1-thread-7] INFO  i.g.d.module.log.EventLogger - Creating log file:/home/sdrtrunk/SDRTrunk/event_logs/20200612_100901.463_772093750_Hz_TRAFFIC_call_events.log  [3GB/4GB 96%]
20200612 100901.464 [sdrtrunk pool-1-thread-7] INFO  i.g.d.module.log.EventLogger - Creating log file:/home/sdrtrunk/SDRTrunk/event_logs/20200612_100901.464_772093750_Hz_TRAFFIC_decoded_messages.log  [3GB/4GB 96%]
20200612 100903.558 [sdrtrunk pool-1-thread-1] INFO  i.g.d.module.log.EventLogger - Creating log file:/home/sdrtrunk/SDRTrunk/event_logs/20200612_100903.558_769768750_Hz_TRAFFIC_call_events.log  [3GB/4GB 98%]
20200612 100903.559 [sdrtrunk pool-1-thread-1] INFO  i.g.d.module.log.EventLogger - Creating log file:/home/sdrtrunk/SDRTrunk/event_logs/20200612_100903.559_769768750_Hz_TRAFFIC_decoded_messages.log  [3GB/4GB 98%]
20200612 100935.453 [sdrtrunk pool-1-thread-4] ERROR i.g.d.d.f.c.PolyphaseChannelManager - Error  [3GB/4GB 99%]
java.lang.OutOfMemoryError: Java heap space
20200612 100939.456 [NioProcessor-518] INFO  i.g.d.a.b.AudioStreamingBroadcaster - [spaz] status: Disconnected  [3GB/4GB 99%]
20200612 101016.224 [sdrtrunk pool-1-thread-5] ERROR i.g.d.d.f.c.PolyphaseChannelManager - Error  [3GB/4GB 99%]
java.lang.OutOfMemoryError: Java heap space
20200612 100946.941 [NioProcessor-498] INFO  i.g.d.a.b.AudioStreamingBroadcaster - [indybay] status: Disconnected  [3GB/4GB 99%]
20200612 101110.506 [NioProcessor-518] WARN  o.a.m.c.f.DefaultIoFilterChain - Unexpected exception from exceptionCaught handler.  [3GB/4GB 97%]
java.lang.OutOfMemoryError: Java heap space
20200612 101128.254 [sdrtrunk pool-1-thread-4] ERROR i.g.d.d.f.c.ContinuousBufferProcessor - Error while dispatching buffers to listener  [3GB/4GB 97%]
java.lang.OutOfMemoryError: Java heap space
20200612 101128.256 [sdrtrunk pool-1-thread-2] INFO  i.g.d.module.log.EventLogger - Creating log file:/home/sdrtrunk/SDRTrunk/event_logs/20200612_101128.256_772093750_Hz_TRAFFIC_call_events.log  [3GB/4GB 97%]
20200612 101128.254 [sdrtrunk pool-1-thread-8] ERROR i.g.d.d.f.c.ContinuousBufferProcessor - Error while dispatching buffers to listener  [3GB/4GB 97%]
java.lang.OutOfMemoryError: Java heap space
20200612 101128.256 [sdrtrunk pool-1-thread-2] INFO  i.g.d.module.log.EventLogger - Creating log file:/home/sdrtrunk/SDRTrunk/event_logs/20200612_101128.256_772093750_Hz_TRAFFIC_decoded_messages.log  [3GB/4GB 97%]
20200612 101128.257 [sdrtrunk pool-1-thread-7] WARN  i.g.d.s.t.u.USBTransferProcessor - [R820T 00000002] - potential memory leak [39 of 40] USB transfer buffers will be disposed during shutdown  [3GB/4GB 97%]
20200612 101219.674 [sdrtrunk pool-1-thread-2] ERROR i.g.d.a.b.AudioStreamingManager - Error processing audio segments for streaming  [3GB/4GB 99%]
java.lang.OutOfMemoryError: Java heap space
20200612 101223.065 [sdrtrunk pool-1-thread-5] ERROR i.g.d.d.f.c.PolyphaseChannelManager - Error  [3GB/4GB 99%]
java.lang.OutOfMemoryError: Java heap space
20200612 101248.883 [sdrtrunk pool-1-thread-5] ERROR i.g.d.d.f.c.ContinuousBufferProcessor - Error while dispatching buffers to listener  [3GB/4GB 97%]
java.lang.IllegalStateException: User count is below zero.  This indicates that this buffer's decrement user count was invoked by more than the expected user count
	at io.github.dsheirer.sample.buffer.AbstractReusableBuffer.recycle(AbstractReusableBuffer.java:119)
	at io.github.dsheirer.sample.buffer.AbstractReusableBuffer.decrementUserCount(AbstractReusableBuffer.java:100)
	at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager$BufferSourceEventMonitor.receive(PolyphaseChannelManager.java:629)
	at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager$BufferSourceEventMonitor.receive(PolyphaseChannelManager.java:577)
	at io.github.dsheirer.dsp.filter.channelizer.ContinuousBufferProcessor.process(ContinuousBufferProcessor.java:189)
	at io.github.dsheirer.dsp.filter.channelizer.ContinuousBufferProcessor$Processor.run(ContinuousBufferProcessor.java:224)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.base/java.util.concurrent.FutureTask.runAndReset(Unknown Source)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.base/java.lang.Thread.run(Unknown Source)
20200612 101248.883 [sdrtrunk pool-1-thread-6] ERROR i.g.d.d.f.c.PolyphaseChannelManager - Error  [3GB/4GB 97%]
java.lang.OutOfMemoryError: Java heap space
20200612 101248.887 [sdrtrunk pool-1-thread-6] ERROR i.g.d.d.f.c.ContinuousBufferProcessor - Error while dispatching buffers to listener  [3GB/4GB 97%]
java.lang.IllegalStateException: User count is below zero.  This indicates that this buffer's decrement user count was invoked by more than the expected user count
	at io.github.dsheirer.sample.buffer.AbstractReusableBuffer.recycle(AbstractReusableBuffer.java:119)
	at io.github.dsheirer.sample.buffer.AbstractReusableBuffer.decrementUserCount(AbstractReusableBuffer.java:100)
	at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager$BufferSourceEventMonitor.receive(PolyphaseChannelManager.java:629)
	at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager$BufferSourceEventMonitor.receive(PolyphaseChannelManager.java:577)
	at io.github.dsheirer.dsp.filter.channelizer.ContinuousBufferProcessor.process(ContinuousBufferProcessor.java:189)
	at io.github.dsheirer.dsp.filter.channelizer.ContinuousBufferProcessor$Processor.run(ContinuousBufferProcessor.java:224)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.base/java.util.concurrent.FutureTask.runAndReset(Unknown Source)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.base/java.lang.Thread.run(Unknown Source)
20200612 101248.889 [sdrtrunk pool-1-thread-8] ERROR i.g.dsheirer.util.ThreadPool - Error while executing runnable in scheduled thread pool [sdrtrunk pool-1-thread-8]  [3GB/4GB 97%]
java.lang.OutOfMemoryError: Java heap space
20200612 101248.904 [sdrtrunk pool-1-thread-2] WARN  i.g.d.s.t.u.USBTransferProcessor - [R820T 00000001] - potential memory leak [39 of 40] USB transfer buffers will be disposed during shutdown  [3GB/4GB 98%]
20200612 101250.232 [NioProcessor-528] INFO  i.g.d.a.b.AudioStreamingBroadcaster - [indybay] status: Connected  [3GB/4GB 99%]
20200612 101301.265 [sdrtrunk pool-1-thread-6] ERROR i.g.d.d.f.c.PolyphaseChannelManager - Error  [3GB/4GB 99%]
java.lang.IllegalStateException: User count is below zero.  This indicates that this buffer's decrement user count was invoked by more than the expected user count
	at io.github.dsheirer.sample.buffer.AbstractReusableBuffer.recycle(AbstractReusableBuffer.java:119)
	at io.github.dsheirer.sample.buffer.AbstractReusableBuffer.decrementUserCount(AbstractReusableBuffer.java:100)
	at io.github.dsheirer.dsp.filter.channelizer.ComplexPolyphaseChannelizerM2.receive(ComplexPolyphaseChannelizerM2.java:234)
	at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager$BufferSourceEventMonitor.receive(PolyphaseChannelManager.java:616)
	at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager$BufferSourceEventMonitor.receive(PolyphaseChannelManager.java:577)
	at io.github.dsheirer.dsp.filter.channelizer.ContinuousBufferProcessor.process(ContinuousBufferProcessor.java:189)
	at io.github.dsheirer.dsp.filter.channelizer.ContinuousBufferProcessor$Processor.run(ContinuousBufferProcessor.java:224)
[... cut ...]
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.base/java.lang.Thread.run(Unknown Source)
20200612 120849.390 [sdrtrunk pool-1-thread-12] ERROR i.g.d.s.t.u.USBTransferProcessor - [R820T 00000000] - error while processing USB transfer buffers  [3GB/4GB 99%]
java.lang.OutOfMemoryError: Java heap space
```

DSheirer, 2020-06-14:
Ken,
Has the software run correctly on this machine in previous versions?

Is the machine robust enough to handle the processing workload?  When Java runs out of memory this quickly, it usually indicates the processor is not keeping up with the procesing workload.

Bobabadabouskie, 2020-06-16:
I have been using the final version of 4.0 and have just started having similar issues. As far as being robust enough...always worked in the past. CPU is only 2 cores @ 2GHz with 8GB ram.

`18:55:19.763 INFO  i.g.d.log.ApplicationLog - Application Log File: /home/max/SDRTrunk/logs/sdrtrunk_app.log  [5MB/124MB 4%]
18:55:19.806 INFO  i.g.d.log.ApplicationLog -   [7MB/124MB 5%]
18:55:19.807 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [7MB/124MB 5%]
18:55:19.807 INFO  i.g.d.log.ApplicationLog - **** sdrtrunk: a trunked radio and digital decoding application ***  [7MB/124MB 5%]
18:55:19.807 INFO  i.g.d.log.ApplicationLog - ****  website: https://github.com/dsheirer/sdrtrunk             ***  [7MB/124MB 5%]
18:55:19.808 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [7MB/124MB 5%]
18:55:19.808 INFO  i.g.d.log.ApplicationLog - Memory Logging Format: [Used/Allocated PercentUsed%]  [7MB/124MB 5%]
18:55:19.808 INFO  i.g.d.log.ApplicationLog - Host OS Name:          Linux  [7MB/124MB 5%]
18:55:19.809 INFO  i.g.d.log.ApplicationLog - Host OS Arch:          amd64  [7MB/124MB 5%]
18:55:19.809 INFO  i.g.d.log.ApplicationLog - Host OS Version:       4.9.0-12-amd64  [7MB/124MB 5%]
18:55:19.810 INFO  i.g.d.log.ApplicationLog - Host CPU Cores:        2  [7MB/124MB 5%]
18:55:19.810 INFO  i.g.d.log.ApplicationLog - Host Max Java Memory:  1 GB  [7MB/124MB 5%]
18:55:19.810 INFO  i.g.d.log.ApplicationLog - Storage Directories:  [7MB/124MB 5%]
18:55:19.810 INFO  i.g.d.log.ApplicationLog -  Application Root: /home/max/SDRTrunk  [7MB/124MB 5%]
18:55:19.811 INFO  i.g.d.log.ApplicationLog -  Application Log:  /home/max/SDRTrunk/logs  [7MB/124MB 5%]
18:55:19.811 INFO  i.g.d.log.ApplicationLog -  Event Log:        /home/max/SDRTrunk/event_logs  [7MB/124MB 5%]
18:55:19.812 INFO  i.g.d.log.ApplicationLog -  Playlist:         /home/max/SDRTrunk/playlist  [7MB/124MB 5%]
18:55:19.812 INFO  i.g.d.log.ApplicationLog -  Recordings:       /home/max/SDRTrunk/recordings  [7MB/124MB 5%]
18:55:20.023 INFO  i.g.d.util.ThreadPool - Application thread pool created with [2] threads  [10MB/124MB 8%]
18:55:20.023 INFO  i.g.dsheirer.gui.SDRTrunk - Home path: /home/max/SDRTrunk  [10MB/124MB 8%]
18:55:20.028 INFO  i.g.d.p.SystemProperties - SystemProperties - loaded [/home/max/SDRTrunk/SDRTrunk.properties]  [10MB/124MB 8%]
18:55:20.028 INFO  i.g.d.p.SystemProperties - SystemProperties - application properties loaded [/home/max/SDRTrunk/SDRTrunk.properties]  [10MB/124MB 8%]
18:55:20.279 INFO  i.g.d.s.SettingsManager - SettingsManager - loading settings file [/home/max/SDRTrunk/settings/settings.xml]  [7MB/27MB 28%]
18:55:20.748 INFO  i.g.d.s.r.RecordingSourceManager - RecordingSourceManager - discovered [0] recording configurations  [10MB/76MB 14%]
18:55:20.774 INFO  i.g.d.s.t.TunerManager - LibUSB API Version: 16777478  [11MB/76MB 15%]
18:55:20.775 INFO  i.g.d.s.t.TunerManager - LibUSB Version: 1.0.22.11312  [11MB/76MB 15%]
18:55:20.776 INFO  i.g.d.s.t.TunerManager - discovered [12] attached USB devices  [11MB/76MB 15%]
18:55:22.396 INFO  i.g.d.s.t.TunerManager - USB Bus [7] Device [0BDA:2838] LOADED: RTL2832 SDR/R820T 00000002 Max Rate:38400000 bps  [13MB/76MB 18%]
18:55:23.801 INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2880000.0] providing [114] channels at [25263.2] Hz each  [14MB/76MB 18%]
18:55:23.936 INFO  i.g.d.s.t.TunerManager - USB Bus [7] Device [0BDA:2838] LOADED: RTL2832 SDR/R820T 00000001 Max Rate:38400000 bps  [14MB/76MB 18%]
18:55:23.938 INFO  i.g.d.s.t.TunerManager - USB Bus [7] Device [2109:2813] Hub Device  [14MB/76MB 18%]
18:55:23.940 INFO  i.g.d.s.t.TunerManager - USB Bus [7] Device [1D6B:0002] Hub Device  [14MB/76MB 18%]
18:55:23.942 INFO  i.g.d.s.t.TunerManager - USB Bus [8] Device [1D6B:0001] Hub Device  [14MB/76MB 18%]
18:55:23.943 INFO  i.g.d.s.t.TunerManager - USB Bus [5] Device [1D6B:0001] Hub Device  [14MB/76MB 18%]
18:55:23.944 INFO  i.g.d.s.t.TunerManager - USB Bus [4] Device [1D6B:0001] Hub Device  [14MB/76MB 18%]
18:55:23.945 INFO  i.g.d.s.t.TunerManager - USB Bus [6] Device [1D6B:0002] Hub Device  [14MB/76MB 18%]
18:55:23.946 INFO  i.g.d.s.t.TunerManager - USB Bus [3] Device [1D6B:0001] Hub Device  [14MB/76MB 18%]
18:55:23.947 INFO  i.g.d.s.t.TunerManager - USB Bus [2] Device [08FF:2810] Unknown Device - Class -1  [14MB/76MB 18%]
18:55:23.948 INFO  i.g.d.s.t.TunerManager - USB Bus [2] Device [1D6B:0001] Hub Device  [14MB/76MB 18%]
18:55:23.948 INFO  i.g.d.s.t.TunerManager - USB Bus [1] Device [1D6B:0001] Hub Device  [14MB/76MB 18%]
18:55:23.949 INFO  i.g.d.s.t.TunerManager - -------------------------------------------------------------  [14MB/76MB 18%]
18:55:23.949 INFO  i.g.d.s.t.TunerManager - USB Bus - Potential Maximum Data Rates  [14MB/76MB 18%]
18:55:23.950 INFO  i.g.d.s.t.TunerManager - USB Bus [7] Rate [76800000] bits per second  [14MB/76MB 18%]
18:55:25.124 INFO  i.g.d.icon.IconManager - loading icons file [/home/max/SDRTrunk/settings/icons.xml]  [11MB/76MB 14%]
18:55:25.554 INFO  i.g.d.p.PlaylistManager - Loading playlist file [/home/max/SDRTrunk/playlist/default.xml]  [21MB/44MB 48%]
18:55:26.327 INFO  i.g.dsheirer.gui.SDRTrunk - starting main application gui  [23MB/44MB 53%]
18:55:26.717 INFO  i.g.d.a.b.AudioBroadcaster - [scanner] status: Connected  [28MB/44MB 65%]
18:55:37.504 INFO  i.g.d.c.c.ChannelAutoStartFrame - Starting [1] channels now - user invoked  [25MB/53MB 48%]
[... cut ...]
19:02:13.481 ERROR i.g.d.s.t.c.TunerChannelSource - Error while sending heartbeat  [1GB/1GB 99%]
java.lang.OutOfMemoryError: Java heap space
19:02:13.481 ERROR i.g.d.s.t.u.USBTransferProcessor - [R820T 00000001] - error while processing USB transfer buffers  [1GB/1GB 99%]
java.lang.OutOfMemoryError: Java heap space

Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "sdrtrunk pool-1-thread-2"

Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "sdrtrunk pool-1-thread-1"`

Bobabadabouskie, 2020-06-16:
> I have been using the final version of 4.0 and have just started having similar issues. As far as being robust enough...always worked in the past. CPU is only 2 cores @ 2GHz with 8GB ram.
> 
> `18:55:19.763 INFO i.g.d.log.ApplicationLog - Application Log File: /home/max/SDRTrunk/logs/sdrtrunk_app.log [5MB/124MB 4%]
> 18:55:19.806 INFO i.g.d.log.ApplicationLog - [7MB/124MB 5%]
> 18:55:19.807 INFO i.g.d.log.ApplicationLog - ******************************************************************* [7MB/124MB 5%]
> 18:55:19.807 INFO i.g.d.log.ApplicationLog - **** sdrtrunk: a trunked radio and digital decoding application *** [7MB/124MB 5%]
> 18:55:19.807 INFO i.g.d.log.ApplicationLog - **** website: https://github.com/dsheirer/sdrtrunk *** [7MB/124MB 5%]
> 18:55:19.808 INFO i.g.d.log.ApplicationLog - ******************************************************************* [7MB/124MB 5%]
> 18:55:19.808 INFO i.g.d.log.ApplicationLog - Memory Logging Format: [Used/Allocated PercentUsed%] [7MB/124MB 5%]
> 18:55:19.808 INFO i.g.d.log.ApplicationLog - Host OS Name: Linux [7MB/124MB 5%]
> 18:55:19.809 INFO i.g.d.log.ApplicationLog - Host OS Arch: amd64 [7MB/124MB 5%]
> 18:55:19.809 INFO i.g.d.log.ApplicationLog - Host OS Version: 4.9.0-12-amd64 [7MB/124MB 5%]
> 18:55:19.810 INFO i.g.d.log.ApplicationLog - Host CPU Cores: 2 [7MB/124MB 5%]
> 18:55:19.810 INFO i.g.d.log.ApplicationLog - Host Max Java Memory: 1 GB [7MB/124MB 5%]
> 18:55:19.810 INFO i.g.d.log.ApplicationLog - Storage Directories: [7MB/124MB 5%]
> 18:55:19.810 INFO i.g.d.log.ApplicationLog - Application Root: /home/max/SDRTrunk [7MB/124MB 5%]
> 18:55:19.811 INFO i.g.d.log.ApplicationLog - Application Log: /home/max/SDRTrunk/logs [7MB/124MB 5%]
> 18:55:19.811 INFO i.g.d.log.ApplicationLog - Event Log: /home/max/SDRTrunk/event_logs [7MB/124MB 5%]
> 18:55:19.812 INFO i.g.d.log.ApplicationLog - Playlist: /home/max/SDRTrunk/playlist [7MB/124MB 5%]
> 18:55:19.812 INFO i.g.d.log.ApplicationLog - Recordings: /home/max/SDRTrunk/recordings [7MB/124MB 5%]
> 18:55:20.023 INFO i.g.d.util.ThreadPool - Application thread pool created with [2] threads [10MB/124MB 8%]
> 18:55:20.023 INFO i.g.dsheirer.gui.SDRTrunk - Home path: /home/max/SDRTrunk [10MB/124MB 8%]
> 18:55:20.028 INFO i.g.d.p.SystemProperties - SystemProperties - loaded [/home/max/SDRTrunk/SDRTrunk.properties] [10MB/124MB 8%]
> 18:55:20.028 INFO i.g.d.p.SystemProperties - SystemProperties - application properties loaded [/home/max/SDRTrunk/SDRTrunk.properties] [10MB/124MB 8%]
> 18:55:20.279 INFO i.g.d.s.SettingsManager - SettingsManager - loading settings file [/home/max/SDRTrunk/settings/settings.xml] [7MB/27MB 28%]
> 18:55:20.748 INFO i.g.d.s.r.RecordingSourceManager - RecordingSourceManager - discovered [0] recording configurations [10MB/76MB 14%]
> 18:55:20.774 INFO i.g.d.s.t.TunerManager - LibUSB API Version: 16777478 [11MB/76MB 15%]
> 18:55:20.775 INFO i.g.d.s.t.TunerManager - LibUSB Version: 1.0.22.11312 [11MB/76MB 15%]
> 18:55:20.776 INFO i.g.d.s.t.TunerManager - discovered [12] attached USB devices [11MB/76MB 15%]
> 18:55:22.396 INFO i.g.d.s.t.TunerManager - USB Bus [7] Device [0BDA:2838] LOADED: RTL2832 SDR/R820T 00000002 Max Rate:38400000 bps [13MB/76MB 18%]
> 18:55:23.801 INFO i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2880000.0] providing [114] channels at [25263.2] Hz each [14MB/76MB 18%]
> 18:55:23.936 INFO i.g.d.s.t.TunerManager - USB Bus [7] Device [0BDA:2838] LOADED: RTL2832 SDR/R820T 00000001 Max Rate:38400000 bps [14MB/76MB 18%]
> 18:55:23.938 INFO i.g.d.s.t.TunerManager - USB Bus [7] Device [2109:2813] Hub Device [14MB/76MB 18%]
> 18:55:23.940 INFO i.g.d.s.t.TunerManager - USB Bus [7] Device [1D6B:0002] Hub Device [14MB/76MB 18%]
> 18:55:23.942 INFO i.g.d.s.t.TunerManager - USB Bus [8] Device [1D6B:0001] Hub Device [14MB/76MB 18%]
> 18:55:23.943 INFO i.g.d.s.t.TunerManager - USB Bus [5] Device [1D6B:0001] Hub Device [14MB/76MB 18%]
> 18:55:23.944 INFO i.g.d.s.t.TunerManager - USB Bus [4] Device [1D6B:0001] Hub Device [14MB/76MB 18%]
> 18:55:23.945 INFO i.g.d.s.t.TunerManager - USB Bus [6] Device [1D6B:0002] Hub Device [14MB/76MB 18%]
> 18:55:23.946 INFO i.g.d.s.t.TunerManager - USB Bus [3] Device [1D6B:0001] Hub Device [14MB/76MB 18%]
> 18:55:23.947 INFO i.g.d.s.t.TunerManager - USB Bus [2] Device [08FF:2810] Unknown Device - Class -1 [14MB/76MB 18%]
> 18:55:23.948 INFO i.g.d.s.t.TunerManager - USB Bus [2] Device [1D6B:0001] Hub Device [14MB/76MB 18%]
> 18:55:23.948 INFO i.g.d.s.t.TunerManager - USB Bus [1] Device [1D6B:0001] Hub Device [14MB/76MB 18%]
> 18:55:23.949 INFO i.g.d.s.t.TunerManager - ------------------------------------------------------------- [14MB/76MB 18%]
> 18:55:23.949 INFO i.g.d.s.t.TunerManager - USB Bus - Potential Maximum Data Rates [14MB/76MB 18%]
> 18:55:23.950 INFO i.g.d.s.t.TunerManager - USB Bus [7] Rate [76800000] bits per second [14MB/76MB 18%]
> 18:55:25.124 INFO i.g.d.icon.IconManager - loading icons file [/home/max/SDRTrunk/settings/icons.xml] [11MB/76MB 14%]
> 18:55:25.554 INFO i.g.d.p.PlaylistManager - Loading playlist file [/home/max/SDRTrunk/playlist/default.xml] [21MB/44MB 48%]
> 18:55:26.327 INFO i.g.dsheirer.gui.SDRTrunk - starting main application gui [23MB/44MB 53%]
> 18:55:26.717 INFO i.g.d.a.b.AudioBroadcaster - [scanner] status: Connected [28MB/44MB 65%]
> 18:55:37.504 INFO i.g.d.c.c.ChannelAutoStartFrame - Starting [1] channels now - user invoked [25MB/53MB 48%]
[... cut ...]
> 19:02:13.481 ERROR i.g.d.s.t.u.USBTransferProcessor - [R820T 00000001] - error while processing USB transfer buffers [1GB/1GB 99%]
> java.lang.OutOfMemoryError: Java heap space
> 
> Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "sdrtrunk pool-1-thread-2"
> 
> Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "sdrtrunk pool-1-thread-1"`

my issue seems to go away when I use only one dongle. But then I get no trunk. I dont know if that means the program will run better/faster.

e850205, 2020-06-16:
Same issue here when using two tuners.

kenrestivo, 2020-06-17:
That machine is an i7 with hyperthreading so 8 cores. And 3 RTL dongles. And a very busy trunking radio system.

closed 2022-11-26 by DSheirer