DSheirer/sdrtrunk #871: Monitor both sides of full-duplex MPT1327 traffic

Flupsy, 2020-06-16:
There's an MPT1327 network near me that uses full-duplex audio, i.e. two traffic channels come up during a call.  It would be nice if sdrtrunk could be configured to bring up both channels so that both sides of the conversation can be heard.

Cagier, 2022-04-19:
Hopefully the NBFM Squelch Feature in #36 will be of assistance here.  We could set up all the Mobile frequencies as NBFM but that wouldn't be very elegant.  Ideally we would be able to get something like the Base on the Left Channel and the Mobile (using a separate Channel Map?) on the Right Channel.  Even separate instances of SDRTrunk with different channel maps could work but again the main blocker is the lack of squelch on an MPT1327 channel (as opposed to an NBFM channel).

Cagier, 2025-11-03:
I've worked out that I may be able to scan all the input frequencies with a separate dongle and then maybe somehow merge the audio files with ffmpeg based on the timestamps but it seems extremely messy as it would not know which call to attach the audio too if there is more than one.  Ideally the Base audio would be on one channel and the mobile would be on the other in the .mp3 file