DSheirer/sdrtrunk #877: Imported channels from RR are offset in spectrum monitor

rstevens011, 2020-06-28:
This could be me being a noob here, but I imported KCERS (A mixed mode P25 Phase 1 system in the Seattle, WA area) and set it up with [Site 1](https://www.radioreference.com/apps/db/?siteId=733) and the spectrum monitor looks weird to me:

[image]

LCN 8 (On 853.175) is the primary control channel for the site in question, but the width of the channel is not wide enough to "encompass" (for lack of a better term) the entire channel and seem to be offset to some degree/  I suspect that is why I am getting all SYNC LOSS errors:

[image]

Is this expected? Attached is my playlist file

DSheirer, 2020-06-28:
Go to the tuners tab, select your tuner and then use the PPM adjustment to align the signals to the frequency readout.

Once you've updated the PPM, the signals should start decoding for you.

rstevens011, 2020-06-28:
@DSheirer thanks for this bit.  I do indeed now have the channels aligned to the frequency/waterfall readout.  When configured with all of the channels listed in the radio reference link above, I do see messages other than SYNC LOSS, so that is a start.  

I however am not getting any audio at all.  I was able to verify using gqrx that I am able to pick up the primary control channel successfully.  I therefore limited the channel config in the playlist file to just the primary and alt control channels thinking that only the control channels should be imported as channels in the config:

[image]

I messed around with the max traffic channels and switched from C4FM and LSM and still no luck.  I also added a second tuner as well:

[image]

[image]

Based on the channel config screenshot above and it being in auto-play mode, I was expecting at least one of these tuners tuned to the control channel on 853.175 but neither are.  I am still getting SYNC LOSS messages repeatedly.  Reading past issues people have suggested setting up a second tuner (despite all of the of these channels being fairly close together within 20Mhz), however with a second tuner,  I am also not picking up any audio either.

DSheirer, 2020-06-29:
The system that you're trying to monitor is a hybrid Motorola SmartZone system with P25 CAI traffic channels.

sdrtrunk does not decode Motorola SmartZone control channels.  It does however decode the P25 traffic channels.  So, don't setup a channel configuration for the control channel.  Setup 1 channel configuration for each of the traffic channels.

When you import this system from radio reference, the playlist editor will setup this sytem automatically for you and it will create a P25 channel configuration for each of the traffic channels.

closed 2020-07-25 by DSheirer