DSheirer/sdrtrunk #886: Call De-Duplication Feature May Be Causing Memory Leak

DSheirer, 2020-07-18:
Several users have reported application running out of memory in version 0.5.0 alpha 2 related to the new call de-duplication feature, specifically with MP3 call recording enabled.

NoahWL, 2020-07-19:
Haven't looked too much into it and I'm still figuring out the processing chains, but this may be related?
<details>
  <summary>AudioRecordingManager.processAudioSegments()</summary>
  
```
private void processAudioSegments()
    {
        RecordFormat recordFormat = mUserPreferences.getRecordPreference().getAudioRecordFormat();
        AudioSegment audioSegment = mCompletedAudioSegmentQueue.poll();

        while(audioSegment != null)
        {
            if(audioSegment.isDuplicate() && mUserPreferences.getDuplicateCallDetectionPreference().isDuplicateRecordingSuppressionEnabled())
            {
                audioSegment.decrementConsumerCount();
            }
            else
            {
                Path path = getAudioRecordingPath(audioSegment.getIdentifierCollection(), recordFormat);

                try
                {
                    AudioSegmentRecorder.record(audioSegment, path, recordFormat);
                }
                catch(IOException ioe)
                {
                    mLog.error("Error recording audio segment to [" + path.toString() + "]");
                }

                audioSegment.decrementConsumerCount();

                //Grab the next one to record
                audioSegment = mCompletedAudioSegmentQueue.poll();
            }
        }
    }
```
</details>

Shouldn't `audioSegment = mCompletedAudioSegmentQueue.poll();` be outside the if statement?  Otherwise it gets stuck in an infinite loop whenever the AudioSegment is flagged as duplicate and the queue continues to fill.

DSheirer, 2020-07-25:
@NoahWL great find, Thank you.  This would explain what folks have been reporting about memory leaks and issues with recording.

closed 2020-07-25 by DSheirer