DSheirer/sdrtrunk #888: Error retrieving country information from radio reference

fofeResonance, 2020-07-22:
Hi,
There was a previous issue for this that was solved with just trying several times, but that's not working for me. I'm having trouble getting the frequency information from the RR source. I'm getting this error each time with differing levels of completion.

```
12:39:15.136 ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [74MB/154MB 48%]
12:39:26.254 ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [101MB/154MB 66%]
12:39:40.714 ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [118MB/154MB 77%]
12:39:52.449 ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [95MB/154MB 62%]
12:39:55.680 ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [65MB/154MB 42%]
12:40:10.563 ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [120MB/154MB 78%]
12:40:13.754 ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [81MB/154MB 52%]
12:46:23.063 ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [126MB/154MB 82%]
12:46:25.798 ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [79MB/154MB 51%]
12:46:56.478 ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [103MB/154MB 67%]
12:47:04.315 ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [94MB/154MB 61%]
12:47:10.346 ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [71MB/154MB 46%]
12:47:13.293 ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [96MB/154MB 62%]
12:47:16.037 ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [120MB/154MB 78%]
12:47:47.758 ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [73MB/154MB 48%]
12:47:51.612 ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [106MB/154MB 69%]
12:49:25.753 ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [95MB/154MB 61%]
12:49:28.954 ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [122MB/154MB 79%]
12:51:50.305 ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [119MB/154MB 77%]
12:51:53.440 ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [75MB/154MB 48%]
12:53:46.610 ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [88MB/154MB 57%]
12:53:50.196 ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [124MB/154MB 80%]
12:55:27.703 ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [76MB/154MB 49%]
12:55:31.624 ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [108MB/154MB 70%]
12:55:39.744 ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [102MB/154MB 66%]
```

Thanks!
Rob

fofeResonance, 2020-07-22:
And it doesn't seem like their service is down ATM.

DSheirer, 2020-07-25:
@fofeResonance Are you still having this issue?

mzac, 2020-07-26:
I'm having the same issue... I'm wondering if somehow my pihole is blocking the dns entry?  I can do a packet capture to see what is going on.

```
11:29:04.504 ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [102MB/331MB 30%]
11:29:28.688 ERROR i.g.d.g.p.r.RadioReferenceEditor - Error retrieving country information from radio reference - Error  [229MB/504MB 45%]
```

mzac, 2020-07-26:
Doesn't look like a DNS issue.  I did a packet capture and see the connection to the Radio Reference API however I don't have a subscription with Radio Reference, could that be the issue?  In the response I get an expiry date: 

```<subExpireDate>12-31-1969</subExpireDate>```

DSheirer, 2020-07-26:
Yes, that's an issue.  You need a premium subscription to access the RR data service.

mzac, 2020-07-26:
> Yes, that's an issue. You need a premium subscription to access the RR data service.

Doh... thanks...

fofeResonance, 2020-07-26:
Ah, makes sense. Thanks for the help!

fofeResonance, 2020-07-26:
I set up for a premium amount and it's working perfectly!
Thanks!

DSheirer, 2020-07-27:
Very good.  I'll close this issue then.

closed 2020-07-27 by DSheirer