DSheirer/sdrtrunk #889: Choppy Audio on Windows

dtickell, 2020-07-23:
Using the same RTL SDR device, identical playlist.xml files, and versions I am able to decode on linux using a virtual machine and the audio is legible. On windows however the audio is choppy and only small chunks are audible. 

Also using the same JMBE library version on both the VM and the native windows machine.

DSheirer, 2020-07-25:
@dtickell Normally, choppy audio can result from bad decodes or from an underpowered CPU.

Can you check if the tuner PPM is adjusted correctly on the Windows box and ensure that you're getting good/consistent decodes by looking at the messages for the audio channel and make sure you're not seeing sync loss messages or CRC errors.

dtickell, 2020-07-25:
So wasnt a CPU issue but at some stage I must have expanded the sample rate
on the tuner on the windows instance and that was the issue.

Thanks for responding and it definitely pointed me in the right direction.