DSheirer/sdrtrunk #902: Feature request: Ability to assign talkgroups to a "preferred" channel

GTR8000, 2020-08-03:
Duplicate Call Detection enhancement: Add an option in the alias editor to assign talkgroups to a "preferred" Channel

Use case: When monitoring two sites of the same system, where one site comes in stronger than the other, it would be beneficial to steer talkgroups to the stronger site whenever possible. When the talkgroup is only active on the weaker site, it will obviously default to that site, however if active on both sites this feature would effectively increase the chances that the best audio is chosen.

GTR8000, 2020-08-03:
Or perhaps an easier method of accomplish the same would be to prefer an entire channel over others when it comes to duplicate call detection. Perhaps an option on the channel page "Prefer this channel over others when detecting duplicate calls" would suffice, rather than having to assign a preference to each talkgroup.

- TG 100 is simultaneously active on Sites 1 and 2, with Site 1 having the stronger RF of the two sites. Site 1 is flagged as the "preferred channel", and therefore duplicate call detection favors the audio from Site 1 over the audio from Site 2.

DSheirer, 2020-08-06:
@GTR8000 I think your second comment is something that could be added easily.  I can add a flag to the channel configuration that allows a user to designate a channel as 'preferred' and that flag can be used to select audio as preferred during the deduplication process.

GTR8000, 2020-08-06:
Thanks, Denny!

After thinking about my first comment some more, it seemed unnecessarily granular to do at the talkgroup level, as well as kind of pointless. If one site/channel is stronger than the other, then every talkgroup at that site is going to presumably have better audio.

I have a followup question that just occurred to me: Does **channel** always = **site**? In other words, if SDRTrunk is setup with two P25 sites as separate channel definitions for the control channels, are the temporary traffic channels automatically associated with the control channel definition for each site?

I suppose what I'm asking is whether or not the temporarily created traffic channels will inherit the control channel's "preferred" status. They would have to in order to ensure that the stronger traffic channels are chosen during deduplication.

DSheirer, 2020-08-06:
Yes.  When you create a channel configuration for a site, all of the traffic channel grants inherit that channel configuration.  So, if the 'preferred' flag is part of the channel configuration then any of the audio created by any of the traffic channels can be made to inherit that 'preferred' flag.

GTR8000, 2020-08-06:
Awesome! This feature will be very helpful, I look forward to it. Thanks again.