DSheirer/sdrtrunk #903: Issue Retrieving Trunked System Setup from RadioReference

bhall7, 2020-08-05:
Hello,

I'm running sdrtrunk v0.5.0-alpha3 and I am trying to retrieve the information from a trunked system from RadioReference, but getting an error, "Radio Reference Error or Service is Currently Unavailable". Other trunked sites come through fine.
[image]

Checking the log shows the following entries that may help resolve this:

```
20200804 203705.146 [sdrtrunk pool-1-thread-11] ERROR i.g.d.g.p.r.SystemEditor - Error retrieving system information  [97MB/150MB 64%]
io.github.dsheirer.rrapi.RadioReferenceException: Error deserializing XML response: [SOAP envelope snipped]
	at io.github.dsheirer.rrapi.RadioReferenceService.submitSync(RadioReferenceService.java:1208)
	at io.github.dsheirer.rrapi.RadioReferenceService.getSites(RadioReferenceService.java:456)
	at io.github.dsheirer.service.radioreference.CachingRadioReferenceService.access$1101(CachingRadioReferenceService.java:44)
	at io.github.dsheirer.service.radioreference.CachingRadioReferenceService.getSites(CachingRadioReferenceService.java:203)
	at io.github.dsheirer.gui.playlist.radioreference.SystemEditor.lambda$setSystem$5(SystemEditor.java:282)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.base/java.util.concurrent.FutureTask.run(Unknown Source)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.base/java.lang.Thread.run(Unknown Source)
Caused by: com.fasterxml.jackson.databind.exc.UnrecognizedPropertyException: Unrecognized field "item" (class io.github.dsheirer.rrapi.type.Bandplan), not marked as ignorable (3 known properties: "base", "offset", "spacing"])
 at [Source: (StringReader); line: 1, column: 4282] (through reference chain: io.github.dsheirer.rrapi.response.ResponseEnvelope["Body"]->io.github.dsheirer.rrapi.response.GetSitesResponse["return"]->java.util.ArrayList[2]->io.github.dsheirer.rrapi.type.Site["bandplan"]->io.github.dsheirer.rrapi.type.Bandplan["item"])
	at com.fasterxml.jackson.databind.exc.UnrecognizedPropertyException.from(UnrecognizedPropertyException.java:61)
	at com.fasterxml.jackson.databind.DeserializationContext.handleUnknownProperty(DeserializationContext.java:823)
	at com.fasterxml.jackson.databind.deser.std.StdDeserializer.handleUnknownProperty(StdDeserializer.java:1153)
	at com.fasterxml.jackson.databind.deser.BeanDeserializerBase.handleUnknownProperty(BeanDeserializerBase.java:1589)
	at com.fasterxml.jackson.databind.deser.BeanDeserializerBase.handleUnknownVanilla(BeanDeserializerBase.java:1567)
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.vanillaDeserialize(BeanDeserializer.java:294)
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.deserialize(BeanDeserializer.java:151)
	at com.fasterxml.jackson.databind.deser.impl.MethodProperty.deserializeAndSet(MethodProperty.java:127)
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.vanillaDeserialize(BeanDeserializer.java:288)
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.deserialize(BeanDeserializer.java:151)
	at com.fasterxml.jackson.databind.deser.std.CollectionDeserializer.deserialize(CollectionDeserializer.java:286)
	at com.fasterxml.jackson.databind.deser.std.CollectionDeserializer.deserialize(CollectionDeserializer.java:245)
	at com.fasterxml.jackson.databind.deser.std.CollectionDeserializer.deserialize(CollectionDeserializer.java:27)
	at com.fasterxml.jackson.databind.deser.impl.MethodProperty.deserializeAndSet(MethodProperty.java:127)
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.vanillaDeserialize(BeanDeserializer.java:288)
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.deserialize(BeanDeserializer.java:151)
	at com.fasterxml.jackson.databind.jsontype.impl.AsWrapperTypeDeserializer._deserialize(AsWrapperTypeDeserializer.java:121)
	at com.fasterxml.jackson.databind.jsontype.impl.AsWrapperTypeDeserializer.deserializeTypedFromObject(AsWrapperTypeDeserializer.java:52)
	at com.fasterxml.jackson.databind.deser.BeanDeserializerBase.deserializeWithType(BeanDeserializerBase.java:1178)
	at com.fasterxml.jackson.databind.deser.impl.MethodProperty.deserializeAndSet(MethodProperty.java:136)
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.vanillaDeserialize(BeanDeserializer.java:288)
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.deserialize(BeanDeserializer.java:151)
	at com.fasterxml.jackson.databind.ObjectMapper._readMapAndClose(ObjectMapper.java:4013)
	at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:3004)
	at io.github.dsheirer.rrapi.RadioReferenceService.deserialize(RadioReferenceService.java:1227)
	at io.github.dsheirer.rrapi.RadioReferenceService.submitSync(RadioReferenceService.java:1193)
	... 10 common frames omitted
```
The system in question is [UCA](https://www.radioreference.com/apps/db/?sid=601), and is a multi-site system--perhaps that is causing a parsing error.

Thanks!

GTR8000, 2020-08-05:
The UCA is an analog SmartZone system, which SDRTrunk is not designed to track. I explained in further detail in your thread on RR.

bhall7, 2020-08-05:
Thanks for the clarification. Closing issue.

DSheirer, 2020-08-06:
@bhall7 I'm going to reopen this issue.

There's a bug hidden in the error stack trace that you posted that has to be addressed in my implementation of the radio reference database api.

bhall7, 2020-08-06:
@DSheirer, thanks! What is interesting is that when I select the Utah Communications Agency system from the drop down, it shows a green check mark next to it, implying that it is a compatible system that SDRTrunk can read. Maybe take a look at that as well. Would love to see legacy Motorola SmartZone analog trunked support in SDRTrunk!

GTR8000, 2020-08-06:
The green check mark is shown because SDRTrunk can import SmartZone traffic channels, however it will not decode the control channel and therefore will not track the system in the traditional manner. Instead you can create a channel definition with all site frequencies, which SDRTrunk will monitor for P25 CAI (i.e. digital) talkgroup activity, and will decode those talkgroups.

Denny can correct me if I'm wrong, but I don't believe SDRTrunk will do anything with the analog audio of a SmartZone system.

DSheirer, 2020-08-07:
Correct.  sdrtrunk can decode each of the P25 CAI traffic channels, but not the analog control channel.

bhall7, 2020-08-07:
Curious. Are there any plans to support traditional, analog, Motorola trunked systems?

PR #923 (2020-09-04): #903 Error Retrieving Radio Reference UCA System
Resolves #903 - updates rrapi library to resolve radio reference web service deserialization issue.
closed 2020-09-04 by DSheirer