DSheirer/sdrtrunk #907: Error starting requested channel - No Tuner Available

Pcpro745, 2020-08-10:
I keep getting this error in the console. It seems that its functioning correctly but curious how to fix.

12:28:11.149 ERROR i.g.d.c.c.ChannelProcessingManager - Error starting requested channel [TRAFFIC:851387500] - No Tuner Available  [425MB/527MB 80%]
12:28:11.549 ERROR i.g.d.c.c.ChannelProcessingManager - Error starting requested channel [TRAFFIC:851387500] - No Tuner Available  [350MB/527MB 66%]
[image]

DSheirer, 2020-08-11:
This means you don't have a tuner available to use for the channel frequency.

closed 2020-08-11 by DSheirer

TRON0903, 2023-06-11:
how do i fix this?sdr trunk shows my sdr but it acts like it aint there when i go to hit play but yet the waterfall see everything.

TRON0903, 2023-06-11:
2023-06-10 23:33:49.913 DEBUG i.g.d.d.f.c.PolyphaseChannelManager - Couldn't design final output low pass filter for polyphase channel source  [172MB/230MB 75%]
2023-06-10 23:33:49.913 ERROR i.g.d.g.p.c.ChannelConfigurationEditor - Error starting channel [Control] - No Tuner Available  [172MB/230MB 75%]
2023-06-10 23:33:54.889 DEBUG i.g.d.d.f.c.PolyphaseChannelManager - Couldn't design final output low pass filter for polyphase channel source  [145MB/230MB 63%]
2023-06-10 23:33:54.890 ERROR i.g.d.g.p.c.ChannelConfigurationEditor - Error starting channel [Control] - No Tuner Available  [145MB/230MB 63%]
2023-06-10 23:35:45.391 DEBUG i.g.d.d.f.c.PolyphaseChannelManager - Couldn't design final output low pass filter for polyphase channel source  [138MB/234MB 59%]
2023-06-10 23:35:45.393 INFO  i.g.d.s.t.m.TunerManager - Unable to source channel [771256250] from preferred tuner [RTL-2832/R820T 00000001] - searching for another tuner  [138MB/234MB 59%]
2023-06-10 23:35:45.393 DEBUG i.g.d.d.f.c.PolyphaseChannelManager - Couldn't design final output low pass filter for polyphase channel source  [138MB/234MB 59%]
2023-06-10 23:35:45.394 ERROR i.g.d.g.p.c.ChannelConfigurationEditor - Error starting channel [Control] - No Tuner Available  [138MB/234MB 59%]

DSheirer, 2023-06-11:
@TRON0903 What is the sample rate that you're using for the tuner?

TRON0903, 2023-06-11:
im very new to all this. Where would i find the sample rate?

TRON0903, 2023-06-11:
[image]

TRON0903, 2023-06-11:
i assume  that is the sample rate in pic

DSheirer, 2023-06-13:
I'm not sure why it's giving you that error.  The error indicates that it can't design the filter for the polyphase channelizer configuration you're using and I'm not sure why because it doesn't look like you're doing anything non-standard.

Try switching from the Polyphase to the Heterdyne channelizer.  Since your tuner appears to be working correctly, this change should let you start a channel.

Go into the User Preferences under the Tuner tab and change to Heterdyne.

TRON0903, 2023-06-13:
[image]

[image]

so i think im making process after switching but im not hearing anything, the right side the frequency is changing but no sound.

TRON0903, 2023-06-13:
[image]