DSheirer/sdrtrunk #913: windows 10 crash

foozfooz75, 2020-08-20:
After running a few hours, I get this. Windows 10 Professional with all the latest, two RTL-SDRs java version "1.8.0_261"

17:45:09.425 INFO  i.g.d.a.c.m.JmbeAudioModule - JMBE audio conversion library loaded: JMBE Audio Conversion Library v1.0.6  [39MB/55MB 71%]
17:45:09.425 INFO  i.g.d.a.c.m.ImbeAudioModule - JMBE audio conversion library IMBE CODEC successfully loaded - P25-1 audio will be available  [39MB/55MB 71%]
17:45:09.447 INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2400000.0] providing [96] channels at [25000.0] Hz each  [40MB/55MB 74%]
17:45:32.706 INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2400000.0] providing [96] channels at [25000.0] Hz each  [70MB/152MB 46%]
17:46:27.937 INFO  i.g.d.s.t.FrequencyErrorCorrectionManager - Auto-Correcting Tuner PPM to [-1.3833976050334889]  [75MB/220MB 34%]
Exception in thread "AWT-EventQueue-0" java.lang.OutOfMemoryError: Java heap space
08:48:33.640 WARN  i.g.d.s.t.u.USBTransferProcessor - [R820T 00000001] - potential memory leak [39 of 40] USB transfer buffers will be disposed during shutdown  [1GB/2GB 99%]
        at java.desktop/sun.java2d.marlin.IntArrayCache.getBuckets(Unknown Source)
        at java.desktop/sun.java2d.marlin.IntArrayCache.getCacheBucket(Unknown Source)
        at java.desktop/sun.java2d.marlin.IntArrayCache$Reference.getArray(Unknown Source)
        at java.desktop/sun.java2d.marlin.IntArrayCache$Reference.widenArray(Unknown Source)
        at java.desktop/sun.java2d.marlin.DRenderer._endRendering(Unknown Source)
        at java.desktop/sun.java2d.marlin.DRenderer.endRendering(Unknown Source)
        at java.desktop/sun.java2d.marlin.MarlinTileGenerator.nextTile(Unknown Source)
        at java.desktop/sun.java2d.pipe.AAShapePipe.renderTiles(Unknown Source)
        at java.desktop/sun.java2d.pipe.AAShapePipe.renderPath(Unknown Source)
        at java.desktop/sun.java2d.pipe.AAShapePipe.fill(Unknown Source)
        at java.desktop/sun.java2d.pipe.PixelToParallelogramConverter.fill(Unknown Source)
        at java.desktop/sun.java2d.SunGraphics2D.fill(Unknown Source)
        at io.github.dsheirer.spectrum.SpectrumPanel.drawSpectrum(SpectrumPanel.java:256)
        at io.github.dsheirer.spectrum.SpectrumPanel.paintComponent(SpectrumPanel.java:170)
        at java.desktop/javax.swing.JComponent.paint(Unknown Source)
        at java.desktop/javax.swing.JComponent.paintChildren(Unknown Source)
        at java.desktop/javax.swing.JComponent.paint(Unknown Source)
        at java.desktop/javax.swing.JLayeredPane.paint(Unknown Source)
        at java.desktop/javax.swing.JComponent.paintToOffscreen(Unknown Source)
        at java.desktop/javax.swing.RepaintManager$PaintManager.paintDoubleBufferedImpl(Unknown Source)
        at java.desktop/javax.swing.RepaintManager$PaintManager.paintDoubleBuffered(Unknown Source)
        at java.desktop/javax.swing.RepaintManager$PaintManager.paint(Unknown Source)
        at java.desktop/javax.swing.RepaintManager.paint(Unknown Source)
        at java.desktop/javax.swing.JComponent._paintImmediately(Unknown Source)
        at java.desktop/javax.swing.JComponent.paintImmediately(Unknown Source)
        at java.desktop/javax.swing.RepaintManager$4.run(Unknown Source)
        at java.desktop/javax.swing.RepaintManager$4.run(Unknown Source)
        at java.base/java.security.AccessController.executePrivileged(Unknown Source)
        at java.base/java.security.AccessController.doPrivileged(Unknown Source)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.desktop/javax.swing.RepaintManager.paintDirtyRegions(Unknown Source)
        at java.desktop/javax.swing.RepaintManager.paintDirtyRegions(Unknown Source)
08:48:43.679 ERROR i.g.d.s.t.c.TunerChannelSource - Error while sending heartbeat  [1GB/2GB 98%]
java.lang.OutOfMemoryError: Java heap space
08:48:43.681 ERROR i.g.d.d.f.c.PolyphaseChannelManager - Error  [1GB/2GB 99%]
java.lang.OutOfMemoryError: Java heap space
08:48:43.682 ERROR i.g.d.d.f.c.ContinuousBufferProcessor - Error while dispatching buffers to listener  [1GB/2GB 99%]
java.lang.OutOfMemoryError: Java heap space
08:48:43.681 ERROR i.g.d.d.f.c.ContinuousBufferProcessor - Error while dispatching buffers to listener  [1GB/2GB 99%]
java.lang.OutOfMemoryError: Java heap space: failed reallocation of scalar replaced objects
08:48:43.702 INFO  i.g.d.s.t.u.USBTransferProcessor - [R820T 00000001] - restart attempt [1]  [1GB/2GB 98%]

Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "sdrtrunk pool-1-thread-7"
08:48:55.454 ERROR i.g.d.d.f.c.ContinuousBufferProcessor - Error while dispatching buffers to listener  [1GB/2GB 98%]
java.lang.OutOfMemoryError: Java heap space
08:48:55.454 ERROR i.g.d.s.t.u.USBMasterProcessor - Error while processing LibUSB timeout events  [1GB/2GB 98%]
java.lang.OutOfMemoryError: Java heap space
08:48:55.458 ERROR i.g.d.util.ThreadPool - Error while executing runnable in scheduled thread pool [sdrtrunk pool-1-thread-5]  [1GB/2GB 98%]
java.lang.OutOfMemoryError: Java heap space
08:49:01.352 ERROR i.g.d.util.ThreadPool - Error while executing runnable in scheduled thread pool [sdrtrunk pool-1-thread-9]  [1GB/2GB 99%]
java.lang.OutOfMemoryError: Java heap space
08:49:01.352 ERROR i.g.d.d.f.c.PolyphaseChannelManager - Error  [1GB/2GB 99%]
java.lang.OutOfMemoryError: Java heap space

Geekhat, 2020-08-31:
Getting this as well...has anyone found a resolution?

DSheirer, 2023-07-14:
Closing as OBE.

closed 2023-07-14 by DSheirer