DSheirer/sdrtrunk #915: Issue with DMR decoder branch

ryanthomp, 2020-08-24:
Getting this error when trying to parse port 4001 UDP packets

20:16:26.215 ERROR i.g.d.d.f.c.o.ChannelOutputProcessor - Error while processing polyphase channel samples  [523MB/1GB 43%]
java.lang.NullPointerException: null
        at io.github.dsheirer.module.decode.ip.udp.UDPPacket.getIdentifiers(UDPPacket.java:108)
        at io.github.dsheirer.module.decode.ip.ipv4.IPV4Packet.getIdentifiers(IPV4Packet.java:130)
        at io.github.dsheirer.module.decode.p25.phase1.message.pdu.packet.PacketMessage.getIdentifiers(PacketMessage.java:271)
        at io.github.dsheirer.alias.action.AliasActionManager.receive(AliasActionManager.java:58)
        at io.github.dsheirer.alias.action.AliasActionManager.receive(AliasActionManager.java:43)
        at io.github.dsheirer.sample.Broadcaster.broadcast(Broadcaster.java:152)
        at io.github.dsheirer.sample.Broadcaster.receive(Broadcaster.java:59)
        at io.github.dsheirer.module.decode.Decoder$MessageDistributor.receive(Decoder.java:103)
        at io.github.dsheirer.module.decode.Decoder$MessageDistributor.receive(Decoder.java:96)
        at io.github.dsheirer.module.decode.p25.phase1.P25P1MessageProcessor.receive(P25P1MessageProcessor.java:95)
        at io.github.dsheirer.module.decode.p25.phase1.P25P1MessageProcessor.receive(P25P1MessageProcessor.java:37)
        at io.github.dsheirer.module.decode.p25.phase1.P25P1MessageFramer.dispatchMessage(P25P1MessageFramer.java:274)
        at io.github.dsheirer.module.decode.p25.phase1.P25P1MessageFramer.receive(P25P1MessageFramer.java:205)
        at io.github.dsheirer.module.decode.p25.phase1.P25P1MessageFramer.receive(P25P1MessageFramer.java:71)
        at io.github.dsheirer.sample.Broadcaster.broadcast(Broadcaster.java:152)
        at io.github.dsheirer.sample.Broadcaster.receive(Broadcaster.java:59)
        at io.github.dsheirer.dsp.psk.PSKDemodulator.broadcast(PSKDemodulator.java:58)
        at io.github.dsheirer.dsp.psk.DQPSKGardnerDemodulator.calculateSymbol(DQPSKGardnerDemodulator.java:88)
        at io.github.dsheirer.dsp.psk.PSKDemodulator.receive(PSKDemodulator.java:115)
        at io.github.dsheirer.dsp.psk.PSKDemodulator.receive(PSKDemodulator.java:89)
        at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderLSM.receive(P25P1DecoderLSM.java:122)
        at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderLSM.receive(P25P1DecoderLSM.java:48)
        at io.github.dsheirer.sample.buffer.ReusableBufferBroadcaster.broadcast(ReusableBufferBroadcaster.java:42)
        at io.github.dsheirer.sample.buffer.ReusableBufferBroadcaster.broadcast(ReusableBufferBroadcaster.java:23)
        at io.github.dsheirer.sample.Broadcaster.receive(Broadcaster.java:59)
        at io.github.dsheirer.sample.buffer.ReusableComplexBufferAssembler.flush(ReusableComplexBufferAssembler.java:158)
        at io.github.dsheirer.sample.buffer.ReusableComplexBufferAssembler.receive(ReusableComplexBufferAssembler.java:132)
        at io.github.dsheirer.sample.buffer.ReusableComplexBufferAssembler.receive(ReusableComplexBufferAssembler.java:144)
        at io.github.dsheirer.dsp.filter.channelizer.output.OneChannelOutputProcessor.process(OneChannelOutputProcessor.java:97)
        at io.github.dsheirer.dsp.filter.channelizer.output.ChannelOutputProcessor.processChannelResults(ChannelOutputProcessor.java:124)
        at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelSource.processSamples(PolyphaseChannelSource.java:248)
        at io.github.dsheirer.source.tuner.channel.TunerChannelSource$ScheduledIntervalProcessor.run(TunerChannelSource.java:321)
        at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515)
        at java.base/java.util.concurrent.FutureTask.runAndReset(FutureTask.java:305)
        at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:305)
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1130)
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:630)
        at java.base/java.lang.Thread.run(Thread.java:832)

Seems that since I cannot give IP addresses aliases it's pulling a null alias given there is no alias for the IP broadcasting cords. Tried adding a broad alias for all Radio IDs still didn't work, only way around has to be either a catch and just to ignore it or in the TODO list I see that IP address aliases are on the way.

maozhenyu123, 2020-08-24:
This branch is still under development and not yet functional.

ryanthomp, 2020-08-24:
Yes I know just wanted to let you guys know of this issue I was talking to Denny about parsing UDP 4001 packets and he said the DMR branch had that capability of doing so I tried it out and found this just wanted to let you guys know it was an issue.

closed 2020-08-24 by ryanthomp