DSheirer/sdrtrunk #928: Error init of VM - 0.5.0 alpha 3 - win7

ah1102, 2020-09-08:
When I try to start the application, an error is displayed via sdr-trunk.bat:

```
ERROR OCCURRED DURING INITIALIZATION OF VM.
Failed setting boot class path.
```

The release of the program is used. Windows 7 x64. Java is not installed.

DSheirer, 2020-09-08:
Make sure you don't have the program running from a directory that is nested too deeply like:

C:\This\Is\A\Reeeeeally\Long\Path\To\where\the\program\is\running\sdrtrunk-release\bin\sdrtrunk.bat.

ah1102, 2020-09-08:
Yes thank you! It works. I have a localized Windows. Perhaps the path was not indicated in English. But the folder with the sdrtrunk program was on a long journey. As you wrote. Moved to c: //

closed 2020-09-08 by ah1102