DSheirer/sdrtrunk #929: Failed to compile DMR on Windows10

Hermite123456, 2020-09-08:
Not sure what I am doing wrong...

**D:\sdrtrunk-5-dmr-decoder-part2>gradlew.bat clean build

FAILURE: Build failed with an exception.

* What went wrong:
java.lang.UnsupportedClassVersionError: org/openjfx/gradle/JavaFXPlugin has been compiled by a more recent version of the Java Runtime (class file version 55.0), this version of the Java Runtime only recognizes class file versions up to 52.0
> org/openjfx/gradle/JavaFXPlugin has been compiled by a more recent version of the Java Runtime (class file version 55.0), this version of the Java Runtime only recognizes class file versions up to 52.0**

java version "1.8.0_261"
Java(TM) SE Runtime Environment (build 1.8.0_261-b12)
Java HotSpot(TM) 64-Bit Server VM (build 25.261-b12, mixed mode)
Java_Home: C:\Program Files\Java\jdk1.8.0_261

[image]

Hermite123456, 2020-09-08:
Please IGNORE, Problem solved

closed 2020-09-08 by Hermite123456

Paulo-D2000, 2022-01-24:
how did you fixed it ?