DSheirer/sdrtrunk #933: SDRPlay Support

Sammy1Am, 2020-09-14:
I know you've mentioned that SDRPlay support is planned of 0.5.0, but I'm interested in digging into it (and maybe just a little impatient 😉) so I've started working on it [in this branch](https://github.com/Sammy1Am/sdrtrunk/tree/sdrplay-support).

I've got the connection to the SDRPlay API working and the device is detected, now I just have to dive into actually controlling it. 
 I'm a big proponent of consistent codebases though, so I understand if it ends up being easier to just re-implement support rather than pull my changes in later.  Optimistically though, I have some questions about the integration that might make a merge later easier, or if nothing else will help me understand the codebase more:

- `initTuners()` in TunerManager is very USB-centric, but SDRPlay (at least for API v3) offers a service for its API instead.  Is the correct place to enumerate and initialize non-USB tuners just in a new `initSDRPlayTuners()` method in TunerManager?
- I'm using JNA to wrap the C api, so there's a small number (4-5) of classes dedicated to this wrapping.  Do you have any preference for where/how those end up in the package?  I've put them in `io.github.dsheirer.source.tuner.sdrplay.api` for now.
- Anything else you'd like me to know to make sure SDRPlay support is unintrusive?  Again, I'll totally understand if it ends up being easier to just redo things rather than merge, but maybe I can be helpful. :)

I also have to figure out how to get `Native.loadLibrary()` to load the `sdrplay_api.dll` from the Program Files folder where it gets installed, but I feel like that's more a generic-Java problem.

DSheirer, 2020-09-16:
The initTuners() method is a good place to place your hooks for adding the sdrplay tuner.  The TunerManager class should evolve to handle any form of tuner.  

Feel free to add/rename methods or reshape the class to balance the needs of devices like sdrplay and the usb tuners.  It probably makes sense to separate the pieces that invoke libusb (e.g. LIBUSB_TRANSFER_PROCESSOR) to only be conditiionally used if/when USB tuners are in use.  It wouldn't make sense to init libusb when only sdrplay tuners are being used.
 
I noticed that one of the usb pieces has leaked into the Tuner class (getMaximumUSBBitsPerSecond).  I'll probably makes sense to insert an abstract USBTuner class into that hierarchy with that method and then update the code in TunerManager to only invoke that method on USB tuner instances.  That's not critical ... just so you don't have to implement that method in your tuner class.

This is outside of the scope of what you're working, but to give you some insight into where I want to eventually go with TunerManager.  I want to refactor all of the USB tuner detection and instantiation off into a USBTunerFactory class.  I also want to add support for runtime unloading of a USB tuner and for blacklisting of a tuner.  I should be able to layer those on top of the changes that you plan to make.

Sammy1Am, 2020-09-16:
Well after a bit of a false start with a tool that auto-generated JNA code, I've redone the wrapper by hand and made some good progress; I'm now receiving data from the SDRplay 👍.  I'm learning a lot as I go, so I'm mostly focusing on getting *some* sort of output routed through sdrtrunk first so that as I go through and clean up my code (and maybe a bit of TunerManager) I'll have a working control case to see if I'm breaking things or not.

I have noticed USB has sort of permeated a lot of the codebase.  For example, it took me a long time to figure out how sdrtuner asks a tuner to `start()` processing data, but eventually figured out that it's in USBTunerController under addBufferListener(), setListener().  (Which actually makes a lot of sense, but took a while to find)

Weirdly, even though getMaximumUSBBitsPerSecond is very USB specific, it still sort of applies to the SDRplay API in this case because the tuner is still a USB device (behind the SDRplay API) and getMaximumUSBBitsPerSecond is only being used (as far as I can tell) to warn about using too much USB bandwidth.  So... I still implemented it, but it feels like it should be optional maybe?  (Or maybe just abstract with a return of 0 as long as it's only being used for that warning)

Speaking of the SDRplay API: it's returning samples in two separate buffers for I and Q.  I'm just now figuring out that I think everywhere else these are just being paired up and buffered into the ReusableComplexBuffer sequentially (I,Q,  I,Q, ...).  So probably it should be trivial to just pair them up inside SDRplay code before sending them out into the sdrtrunker world.... I think.  Are there any consumers that don't separate those out into separate values before working with them?  It would be a big change probably, but I might look at what it would take to make a new ReusableComplexBuffer class that stores both buffers so I don't have to interleave I/Q just to have them separated again later.  -- a project for later maybe, like I said I'm mostly just trying to get *anything* end-to-end working so it'll be easier to see when I break stuff.

I've actually put most of the SDRplay detection and instantiation outside TunerManager already, so I'll go ahead and name that SDRplayTunerFactory so it's ready for the future.

Sammy1Am, 2020-09-16:
Actually, I have run into a bit of a conundrum/bug? (when you have some free time, I've got a workaround for now):

There are [two `reloadFrom()` methods in ReusableFloatBuffer](https://github.com/DSheirer/sdrtrunk/blob/master/src/main/java/io/github/dsheirer/sample/buffer/ReusableFloatBuffer.java#L87-L126), but only one of them checks `mUserCount` before reloading the buffer.  [`getBuffer()` from `ReusableComplexBufferQueue`](https://github.com/DSheirer/sdrtrunk/blob/b061db11893d412aab9be16467d9f41662dc8d6f/src/main/java/io/github/dsheirer/sample/buffer/ReusableComplexBufferQueue.java#L48) automatically increments the userCount as soon as it creates a new buffer.

This `incrementUserCount()` is clearly needed somewhere because if it weren't decremented evenly there would be issues.  However, this also means that it's not ever possible to call `getBuffer` followed by `reloadFrom(float[] samples...)` because it will throw an exception.

But that's exactly what's happening for USB tuners [in the `NativeBufferConverter` class](https://github.com/DSheirer/sdrtrunk/blob/master/src/main/java/io/github/dsheirer/source/tuner/usb/converter/NativeBufferConverter.java#L48-L51) except because it's using `FloatBuffer` instead of `float[]` there's no check and no exception.

It seems like either `ReusableComplexBufferQueue` shouldn't be incrementing the userCount for new queues (who's using it if it's brand new?), or there needs to be a way to load the queue before it's returned with already 1 userCount.  For now I'm just using `FloatBuffer` instead as a workaround, but something seems off.

DSheirer, 2020-09-17:
It looks like I've been violating the integrity of the ReusableComplexBuffer and simply accessing the internal float[] and writing the values directly to the buffer's float array.  I think I did this to avoid creating a temporary copy of a float[] that would then just have to be copied back into the reusable buffer.

It looks like you could copy the ReusableBufferQueue.getBuffer(float[], timestamp) method and paste it into the ReusableComplexBufferQueue class to give the method that you're looking for, of course changing it to return a complex buffer versus a standard buffer.

The thought process behind auto-incrementing the user count when you get a buffer from the queue was that you don't get a buffer unless you intend to send it to someone and to ensure that the user count got set correctly without having to remember

Sammy1Am, 2020-10-06:
An update on progress:

It took nearly 3 weeks to finally wrap the SDRplay API in Java; I unsuccessfully tried JNA, JNI, SWIG, and Bridj before finally struggling through the nigh-undocumented JNR-FFI to create a wrapper.  Given the complexity of the wrapper, I've created it as a separate library [in this repository](https://github.com/Sammy1Am/SDRplayJava) and am just using Gradle to import it into sdrtrunk locally.  I'll try to get that repo published to mavenCentral or somewhere once I've got things stable so it can be easily pulled in.

I just now successfully tuned to a NBFM station, and sort of a P25 control channel using the code on [this branch](https://github.com/Sammy1Am/sdrtrunk/tree/sdrplay-support).  There's some weirdness going on (like the waterfall shows 2MHz of bandwidth, but there are only discernable signals in the center 1/8 of the display), but I think that's mostly up to my lack of SDR knowledge.

There are still some SDRplay features I need to add (LNA, gain, etc.), and I may take a stab at some of the buffer changes you mentioned above.  I'm eager to make sure this code is available to other more knowledgeable SDR coders who might be able to optimize the SDRplay support in ways I can't yet fathom; at what point do you think it makes sense to try to put together a pull request?  At a minimum I should get the Java wrapper into mavenCentral, but after that is a pull request for bare-minimum functionality worth it, or should I try to invite contributions to my branch separately until the SDRplay support is a little more fleshed out?

alphafox02, 2020-10-06:
This is great news and thank you for sharing over at the SDRangel site as well. I’d like to try what you’ve done with sdrtrunk. I have a rsp1a and rspduo.

Sammy1Am, 2020-10-07:
> I’d like to try what you’ve done with sdrtrunk.

Ideally if you could just clone the two repos I linked in my last post, you'll just need edit settings.gradle in the sdrtrunk project to point to wherever you cloned SDRplayJava to.  It should be as simple as opening both projects in NetBeans (other IDEs are available), editing the file, and hitting "Run".  I've been running the code on a Windows machine though, so there may be a bit of change for Linux needed if that's what you're on.

If you're not in a position to build the code yourself, I can try to make a packaged .jar file for you to test, with the caveat that the support is still *very* rudimentary (like you can't even change sample rate yet, and parts of it still say "HackRF" where it shouldn't).  I plan to keep working on it this week though, so I should have something a *little* more finished before too long.  I haven't filled in any of the RSPduo code yet, so dual-tuners won't work, but the device itself should.

alphafox02, 2020-10-07:
Great job and thanks for working on this. I’ll try the steps you mentioned and if I can’t get it to work this week maybe I could grab the jar from you and test it out over the weekend.

DSheirer, 2020-10-07:
> I unsuccessfully tried JNA, JNI, SWIG, and Bridj before finally struggling through the nigh-undocumented JNR-FFI to create a wrapper.

@Sammy1Am 
Timing is everything sometimes ... project Panama is scheduled to be released as an incubator module in Java 16 (Mar 2021).  It provides out-of-the-box support for mapping Java to native libraries:

https://inside.java/2020/10/06/jextract/

Sammy1Am, 2020-10-19:
Huh.  That figures 😐.

Well at this point the JNR work's been done, so I think I'll just leave it for now (especially in case any users get grumpy about having to update Java-- I still get people running into issues because they're still on Java 8!).  I have extracted all the wrapping work into a separate SDRplayJava library though and published it to jcenter.  This upshot of this is that the Java API bindings should be relatively wrapper-agnostic, so if it makes sense to switch to Panama later there shouldn't need to be any changes to code in sdr-trunk (only internally in the library).

I have things *relatively* buttoned up on the initial support, so I'm going to work on a pull-request now.  A couple TODOs and details, but I'll details those in the PR.  I don't think I can turn this issue **into** a PR, but I'll make sure I link them.

alphafox02, 2020-11-15:
I have both repos pulled down. I initially thought I did a successful build, but it also seemed way to easy. Turns out when I pull down your sdrplay repo I’m not actually getting the sdrplay relevant files. So I hand edited the latest normal release of sdrtrunk with your changes. Tried to build in Linux with gradlew, but of course it failed I think do to me not having the SDRplayJava repo. 

How do I link to the SDRplayJava directory from the settings.gradle file in the main sdrtrunk directory? Maybe I’m really missing a step or I just am way off in what I’m doing.

Sammy1Am, 2020-11-15:
It sounds like you eventually got it figured out, but just to address this problem (in case anyone else comes across it): make sure you've got the [`sdrplay-support`](https://github.com/Sammy1Am/sdrtrunk/tree/sdrplay-support) branch checked out instead of `master` to make sure you have all the SDRplay changes.

Also, SDRplayJava *should* get pulled in from jCenter now, so you shouldn't need to actually check out the SDRplayJava repo (unless you want to make some changes to it).

neilharvey94044, 2021-01-18:
I've just started looking at sdrtrunk.  I got a download of sdr-trunk-windows-x86_64-v0.5.0-alpha6 and tried running with my RSP1A connected.  No joy, doesn't seem to recognize it.   Wondering if SDRPlay is still in the dev queue or if there is something else I need to do?  I've yet to use sdrtrunk at all, so I'm probably missing something.

I should be receiving an RSP-SDR dongle next week, will also try with that.

alphafox02, 2021-01-18:
You can follow along with this video and get it working in Linux so long as you either use DragonOS or have the sdrplay api installed on whatever Linux you’re using.

DragonOS Focal SDRTrunk P25 Phase 1 w/ SDRPlay support (RSP1A) https://youtu.be/oIlchDwe2P4

But for either windows or Linux I think you’ll still need to be building it all yourself.

dseven, 2021-01-18:
Finally got around to trying this today (with a RSPdx on Windoze 10). It almost works. It's decoding some P25 control messages, and it sees channel grants, but it's not able to decode any audio so-far. I also see a lot of this (not sure if it's normal?):

```
17:19:40.511 INFO  i.g.d.s.t.u.USBMasterProcessor - Stopping USB master processor thread  [1GB/3GB 48%]
17:19:41.902 INFO  i.g.d.s.t.u.USBMasterProcessor - Starting USB master processor thread  [2GB/3GB 65%]
17:19:43.011 INFO  i.g.d.s.t.u.USBMasterProcessor - Stopping USB master processor thread  [1GB/3GB 36%]
17:19:44.020 INFO  i.g.d.s.t.u.USBMasterProcessor - Starting USB master processor thread  [2GB/3GB 55%]
17:19:45.092 INFO  i.g.d.s.t.u.USBMasterProcessor - Stopping USB master processor thread  [2GB/3GB 69%]
17:19:46.101 INFO  i.g.d.s.t.u.USBMasterProcessor - Starting USB master processor thread  [1GB/3GB 27%]
17:19:47.179 INFO  i.g.d.s.t.u.USBMasterProcessor - Stopping USB master processor thread  [1GB/3GB 46%]
17:19:48.187 INFO  i.g.d.s.t.u.USBMasterProcessor - Starting USB master processor thread  [2GB/3GB 62%]
```

and traffic decoded messages just shows like:

```
DECODED Message Logger

20210117 172545,PASSED,<-> SYNC LOSS - BITS PROCESSED [1570]
20210117 172545,PASSED,<-> SYNC LOSS - BITS PROCESSED [1570]
20210117 172545,PASSED,<-> SYNC LOSS - BITS PROCESSED [1570]
20210117 172546,PASSED,<-> SYNC LOSS - BITS PROCESSED [1570]
20210117 172546,PASSED,<-> SYNC LOSS - BITS PROCESSED [1570]
```

Quite a bit of this too:

```
17:27:42.146 INFO  i.g.d.s.t.s.SDRplayTunerController - Power Overload: Overload_Detected  [2GB/3GB 63%]
17:27:42.146 INFO  i.g.d.s.t.s.SDRplayTunerController - Power Overload: Overload_Corrected  [2GB/3GB 64%]
```

HackRF One still works, so I'm confident in my config, other than the SDRplay-specific parts. I've tried sample rate and IF bandwidth of 6M and 8M, and fiddled with IF Mode a bit. I've cranked "RF Gain Reduction" to the max, but it still looks a bit overdriven. May need an extra gain knob for the RSPdx (antenna selector too:)

dseven, 2021-01-18:
I just had a "duh" moment (sortof). I had been leaving the HackRF one (USB) connected as well as the RSPdx, and I just moved the antenna between them. I just noticed that, when I was failing to receive traffic, the HackRF tuner was getting locked and unlocked. Aha! I tried again with the HackRF unplugged, and now I have "all kinds of audio"! FWIW, I did have the "Preferred Tuner" under "Source" set to the SDRplay, but I only have the control channel frequency in there. Maybe I'd need to add all channels.

Anyway, now the main P25 system that I usually monitor seems to be working well. I tried on another system that's a bit marginal in strength, and I'm seeing some weirdness that seems to be related to rolloff at the ends of the sample range and the way that SDRtrunk retunes the center frequency when "trunk-tracking", but I have some issues like this when using the HackRF too (discussion for another day, perhaps).

Nice work! I'll tinker with it some more.

neilharvey94044, 2021-01-22:
I was able to use the video and repeat the build and run steps for Windows 10.  It seems to be working with an SDRPlay RSP1A, but isn't getting much traffic.  I don't have a control scanner to compare it to, so really not sure if it's my radio signal or the build.  It definitely identifies and plays the RSP1A and I got a few calls from local jails and sheriff that came in really well, but not the amount of traffic I would have expected.  Will keep playing with it and see what happens.

Receiving lots and lots of these events in the event log:
20210121 160818,PASSED,<-> SYNC LOSS - BITS PROCESSED [1570]

neilharvey94044, 2021-01-23:
I've tested this same repo from Sammy1am with branch sdrplay-support for both an RSP1A and a RTL-SDR.  The RSP1A is definitely not working, whereas the RTL-SDR is picking up lots of traffic.  So not my antenna setup, definitely something to do with the RSP1A handling in this branch.  My RSP1A works flawlessly with SDRuno, so likely nothing wrong with it.
Unfortunately, I have a relatively good Java background but know very little about the engineering behind these SDR's and radio in general.  I'll poke around more, but not optimistic that I can solve in any reasonable amount of time.

I don't see any word from Sammy1am, so perhaps he's moved on to other things.

alphafox02, 2021-01-23:
Hello. So I’d have to try it out again, but with my rsp1a (like in the video) I was able to receive and listen to the p25 phase 1 audio in my area with the little antenna that comes with the rtlsdr. It was awhile ago, but I think the gain values needed messed with. Might not have been as optimized as the rtlsdr, but it did work about the same as one for me. Hopefully we’ll see some more developments.

neilharvey94044, 2021-01-31:
I watched the video again...woohoo! Got audio on my RSP1A. The settings were critical to getting audio. Using IF_Mode:IF_Zero, IF_Bandwidth:IF_BW_8_00, Sample Rate:8000000. This got the audio going and it seems to be working fine, especially when I put it on the discone I have on the roof.

I'm curious can anyone point me to reference material on the SDRPlay API settings?  Not the API, I have that.  Unfortunately the API assumes you already know what you're after and explains only the basic API usage, not the meaning or result of any settings.  From the settings above, it looks like the Intermediate Frequency Bandwidth needs to match the Sample Rate Bandwidth, but that's just me guessing.  I'm even guessing what the acronyms represent.  Obviously I'm not an EE.

This datasheet for the tuner chip of the RSP1A is the best I could find.
[MSi001 Datasheet R3P4.pdf](https://github.com/DSheirer/sdrtrunk/files/5899002/MSi001.Datasheet.R3P4.pdf)

theGuruWithin, 2021-03-03:
Hi I was wondering does a win7 version of SDRPlay RSP1A version, if so where might I get a copy to try?

in Gratitude

neilharvey94044, 2021-03-03:
Not sure I understand your question.

Are you asking...
"I have an RSP1A device and I want to run sdrtrunk with it on Windows 7?"

I believe Windows 7 is 64 bit, so yes you should be able to run it.  You will probably need to build it, however.  You will need to install git, Java JDK 13 or 15, and Gradle in order to build and run it.

The following will build the latest version (pull request) that supports SDRPlay devices.

To build:
git clone  https://github.com/DSheirer/sdrtrunk.git
cd sdrtrunk
git fetch origin pull/1040/head:sdrplay-support
git checkout sdrplay-support
./gradlew build

Distributions will be in .\build\distributions

theGuruWithin, 2021-03-05:
Hi Neil,

Thank you so much for your quick response.

I have accomplished the following steps and had a few questions on how to proceed to next steps.

To build:
git clone  https://github.com/DSheirer/sdrtrunk.git
cd sdrtrunk
git fetch origin pull/1040/head:sdrplay-support
git checkout sdrplay-support

I am not sure how and where do I install the Java and gradle
(is it possible to install these dependencies from command line)
Java JDK 13 or 15, and Gradle

and then how do I complete process in order to build and run it?  is the ./gradlew build all that is necassary once Java and Gradle are installed?

Thank you for helping a newbee to git as well as building software!!!

tgw

dseven, 2021-03-05:
You don't have to install gradle (or at least I didn't). Install openjdk (I used version 15, but 13 should work too). Install the SDRplay API (v3). You'll need to set the environment variable JAVA_HOME to point to the location of the jdk installation, and you'll need the SDRplay API to be in the PATH environment variable. In PowerShell, I set these with:

```
$env:JAVA_HOME = "C:\otherApps\openjdk\jdk-15"
$env:PATH += ";C:\Program Files\SDRplay\API\x64"
```

Then you can build with `gradlew.bat clean build` and run with `gradlew.bat run`

Sammy1Am, 2021-03-05:
For what it's worth, the NetBeans IDE (which is what I'm familiar with, but I'm sure Eclipse does as well) has a Gradle plugin that will allow you to open Gradle repos as projects.  Following the installation for NetBeans should take care of some of the Java/Gradle installation dependencies for you if you're willing to go GUI instead of command line.

You can [install JDK 15 from here](https://www.oracle.com/java/technologies/javase-jdk15-downloads.html) (might be a prerequisite for NetBeans anyway).  There's some info on installing [Gradle on Windows here](https://gradle.org/install/) as well-- you'll probably need to add it to your PATH.

fjl82, 2021-12-03:
I've just built this branch on Fedora 34 on OpenJDK 17, as per the instructions above, and I think the build went fine (no errors reported). It works with the RTL SDR, as before. But the SDRplay RSP1a does not work well yet. The device is found, a tuner is added, and I can configure this tuner. The SDRplay device is also in use when sdrtrunk is running, as other applications can no longer open it until it is closed, so it is at least doing something. But every attempt to use it results in an error like this:
```
Exception in thread "AWT-EventQueue-0" io.github.sammy1am.sdrplay.ApiException$FailException: sdrplay_api_Fail
	at io.github.sammy1am.sdrplay.ApiException.checkErrorCode(ApiException.java:43)
	at io.github.sammy1am.sdrplay.ApiException.checkErrorCode(ApiException.java:24)
	at io.github.sammy1am.sdrplay.SDRplayDevice.init(SDRplayDevice.java:167)
	at io.github.dsheirer.source.tuner.sdrplay.SDRplayTunerController.addBufferListener(SDRplayTunerController.java:258)
	at io.github.dsheirer.spectrum.SpectralDisplayPanel.showTuner(SpectralDisplayPanel.java:448)
	at io.github.dsheirer.source.tuner.TunerSpectralDisplayManager.receive(TunerSpectralDisplayManager.java:51)
	at io.github.dsheirer.source.tuner.TunerSpectralDisplayManager.receive(TunerSpectralDisplayManager.java:26)
	at io.github.dsheirer.source.tuner.TunerModel.broadcast(TunerModel.java:235)
	at io.github.dsheirer.spectrum.ShowTunerMenuItem$1$1.run(ShowTunerMenuItem.java:55)
```
The RSP1a works fine in GQRX, CubicSDR, SDRangel and several DAB applications, so I'm pretty sure my SDRplay setup is ok. Any idea what's going on and/or how to fix it?

Not sure if it's related, but I also get application lockups in this build, when using the RTL SDR, that I didn't get in the old build. When not tuned to any channels, there are no lockups.

closed 2023-07-14 by DSheirer

jkldgoefgkljefogeg, 2023-12-16:
Thanks, [0.6.0 Beta 5](https://github.com/DSheirer/sdrtrunk/releases/tag/v0.6.0-beta-5) works great with RSP1A

in case it's not obvious

- install SDRplay API https://www.sdrplay.com/api/
- install sdrtrunk 0.6.0+ https://github.com/DSheirer/sdrtrunk/releases